mpromonet/webrtc-streamer #434: Streamer gets 'Core dump' when compiling WebRTC with Debug configuration

ihabZhaika, 2021-07-18:
Hey,
I tried to compile the streamer with 'Release' settings then tried to stream file and it work as it should, now i switched the Libwebrtc build to "Debug" on the CMake file, after compiling the streamer and run it, when I  open a web page for watching the streamed video it get core dump as the following:
[image]

This is the system settings that I am running on:
[image]

To reproduce it follow the following steps:

1. Clone the code, make sure the commit is: 4922c95934b084a0f97a239d838dd36969ee1f99
2. Change the in the CmakeList file from 'Release' to 'Debug', i.e: set (CMAKE_BUILD_TYPE "Debug")
3. Run cmake .
4. Run make
5. Run the following command "./webrtc-streamer -N 4  -n video3 -u rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov"
6. Open the browser on the server IP and the streamer will get Core Dump

Any suggestions to mitigate the issue ?

mpromonet, 2021-07-18:
Hi,

This is a bug, usage of audio stuff doesnot follow webrtc sdk thread access check.
It may works disabling audio wiht `-a` option ?

Best Regards,
Michel.

ihabZhaika, 2021-07-18:
> Hi,
> 
> This is a bug, usage of audio stuff doesnot follow webrtc sdk thread access check.
> It may works disabling audio wiht `-a` option ?
> 
> Best Regards,
> Michel.

It helped with the TRSP video, it do work now with the -a.
For other streams still crashing, see the following image
[image]

mpromonet, 2021-07-18:
Hi,

This seems a very different problem, maybe B-frame bad managed by WebRTC SDK ?

Best Regards,
Michel.

ihabZhaika, 2021-07-18:
> Hi,
> 
> This seems a very different problem, maybe B-frame bad managed by WebRTC SDK ?
> 
> Best Regards,
> Michel.

Not sure, 
I tried the null codec [ -o ], it do work now without crashing but I have to keep those flags -a -o, in addition I keep getting the following
[image]

commit bf2a7f4e (2022-02-12): fix #434: access to audio module from worker thread
closed 2022-02-12 by mpromonet