mpromonet/webrtc-streamer #435: Cannot play local videos. Live demos also appear broken

ndtrek07, 2021-07-29:
After building webrtc-streamer on Ubuntu 18.04 and running with the following command, the web interface never finishes loading the video (on chrome, the spinner continues indefinitely).

`./webrtc-streamer file://example.mov`

The same behavior is observed when attempting to view the "live demo" at https://webrtc-streamer.herokuapp.com/.

mpromonet, 2021-07-29:
Hi,

The live demo seems working, some of the stream are not working time to time.
Stream a file works only with a mkv. I guess you should have some message saying this.

Best Regards,
Michel.

ndtrek07, 2021-07-29:
Thanks for the clarification that mkv is the only supported streaming file type.

Must be something wrong with my ISP. Live demo only works for me when connected over a VPN, otherwise all fail to play.

closed 2021-09-08 by mpromonet