mpromonet/webrtc-streamer #436: Support H265 streaming

vegalou, 2021-08-03:
**Webrtc-Streamer support H265 streaming decode**

At present, more IP Cameras support the H265 RTSP streaming format as a standard, 
and hope that it could be supported as default codec. 

Thanks & Regards.

mpromonet, 2021-08-03:
Hi,

WebRTC SDK doesnot include an H265 decoder, and I would like to avoid others dependencies to keep this project as simple it can. 
However if you like to submit a pull request, I will look to it, you may use ffmpeg for this.
You may also ask WebRTC SDK to include H265 support https://bugs.chromium.org/p/webrtc/issues/entry 

Best Regards,
Michel.

vegalou, 2021-08-03:
Hi @mpromonet , 

I could feed streaming server with ffmpeg decode both H264 & H265, and play video in HTML Video in low latency.

WebRTC get video source from webcam normally(Almost H264 currently), 
Webrtc-Streamer could fetch IPCam's RTSP to HTML5 Video.( New IPCam support H265 in the majority)

Maybe H265 codec would be supported like currently H264 mp4 in common. (Some player needs WASM to play H265 files in web) 

Regards.

closed 2021-08-03 by vegalou

yangjieshao, 2023-03-08:
[https://caniuse.com/?search=h265](https://caniuse.com/?search=h265)
[image]
@mpromonet 
看上去 Chrome从107开始支持H265了
(It seems that Chrome has supported H265 since 107)
是否有相关的计划以支持H265？
(Is there a plan to support H265?)
或者是只需要在浏览器上启动H265硬解码就可以使用播放H265的流了？
(Or just start H265 hard decoding on the browser to play H265 stream?)

vegalou, 2023-03-08:
@yangjieshao 

webrtc支援h265才有機會，webrtc-streamer是根據webrtc為基礎的封裝轉換RTSP， #mpromonet  作者是這樣回應

我的解決方案是用ffmpeg轉入streaming server成為HTML5，所以不管源流RTSP格式都可以跨過。

yangjieshao, 2023-03-08:
> @yangjieshao
> 
> webrtc支援h265才有機會，webrtc-streamer是根據webrtc為基礎的封裝轉換RTSP， #mpromonet 作者是這樣回應
> 
> 我的解決方案是用ffmpeg轉入streaming server成為HTML5，所以不管源流RTSP格式都可以跨過。

所以说 我就是在隔壁看见 似乎webrtc已经支持了 才来问的
 [https://github.com/deepch/RTSPtoWebRTC/issues/134](https://github.com/deepch/RTSPtoWebRTC/issues/134)

[https://github.com/open-webrtc-toolkit/owt-deps-webrtc/blob/deef390463/common_video/h265/h265_common.h](https://github.com/open-webrtc-toolkit/owt-deps-webrtc/blob/deef390463/common_video/h265/h265_common.h)

vegalou, 2023-03-08:
@yangjieshao 

您提的文串沒有作者提及要支援在本專案上。

其實issue裡面作者早就回應我了，你可以根據SRC把h.265啟用，自己編出支援h265的webrtc-streamer

我提的方案已經實做應用了，RTSP(264/265) to HTML5 VIDEO

簡單一點，用一個RTSP Proxy，將H265轉發為H264提供給webrtc-streamer，這也是驗證過的方法。

yangjieshao, 2023-03-08:
@vegalou 
我只是在隔壁看见似乎有支持的可能所以来问一下
在具体项目中，其实是可以接受让业主把摄像头的流改为H264的
感谢回复

mpromonet, 2023-03-08:
Hi,

The webrtc SDK used doesnot support H265 for now, there is this openned issue https://bugs.chromium.org/p/webrtc/issues/detail?id=13485&q=H265&can=2.

Maybe moving to [owt-deps-webrtc](https://github.com/open-webrtc-toolkit/owt-deps-webrtc) is a possible way ?

Best Regards,
Michel.

vegalou, 2024-01-22:
Latest 0.8.5 seems support H265, I'll test it later.

mpromonet, 2024-01-22:
It is probably not working, webrtc H265 for Safari seems a kind of hack....

vegalou, 2024-01-23:
@mpromonet 
Thanks! I'd the solution for H264/H265 RTSP in HTML5 for Browser or Mobile via FFMPEG already.

`webrtc-streamer is an easy way to H264 RTSP. Great Work!!`