mpromonet/webrtc-streamer #437: [1276:438][29132] (livevideosource.h:175): LiveVideoSource: onData no decoder NOTIFY failed

heyikang, 2021-08-04:
**Is your feature request related to a problem? Please describe.**
I'm not sure if this is a bug or a feature. This problem occurred when I connected the camera

**Describe the solution you'd like**
Add a new decoder?

**Describe alternatives you've considered**
no

**Additional context**
System  Windows 10 Pro x64 19042.1110
webrtc-streamer  last release

heyikang, 2021-08-04:
The above problem does not exist in VLC media player, it can be played normally

heyikang, 2021-08-04:
[image]

mpromonet, 2021-08-04:
Hi,

Probably in the logs there is something like format not supported.

Best Regards,
Michel.

heyikang, 2021-08-05:
@mpromonet Thanks for your help, how can I solve this problem

mpromonet, 2021-08-05:
Hi,

I think you have choice between using a supported format (H264, VP9, JPEG) or implementing the decoder for your format.

Best Regards,
Michel.

heyikang, 2021-08-06:
@mpromonet Thanks for your help

closed 2021-08-06 by heyikang