mpromonet/webrtc-streamer #442: 使用默认提供的docker启动方式无法获取到视频（The video cannot be obtained by using the default docker startup method）

lhDream, 2021-08-09:
默认提供的docker启动方式
docker run -p 8000:8000 -it mpromonet/webrtc-streamer
无法获取到rtsp视频流，经过检查发现是启动后的docker容器内部无法使用网络导致

使用主机网络方式启动后可以正常使用
docker run -it --rm --network=host --name=webrtc mpromonet/webrtc-streamer /bin/bash

使用环境：
系统：centos 7 x86_64
docker 版本：Docker version 1.13.1, build 7d71120/1.13.1

-------------------------------------------------------------------机翻----------------------------------------------------------------------
The default docker startup method is provided

docker run -p 8000:8000 -it mpromonet/webrtc-streamer

Unable to obtain RTSP video stream. After inspection, it is found that the network cannot be used inside the docker container after startup

It can be used normally after the host network mode is started

docker run -it --rm --network=host --name=webrtc mpromonet/webrtc-streamer /bin/bash

Operating environment:

System: CentOS 7 x86_ sixty-four

Docker version: docker version 1.13.1, build 7d71120 / 1.13.1

yangjieshao, 2021-08-10:
- -  网不通就别难为人家了

lhDream, 2021-08-10:
该说还是要说的，就算他没时间改，也能让后面的兄弟少踩一些坑

closed 2021-09-20 by mpromonet

ShareQiu1994, 2022-03-22:
该说不说 还是有点帮助的