mpromonet/webrtc-streamer #448: Skip transcode option for each stream, instead of globally

dpd123, 2021-09-06:
This is really amazing software! It has simplified everything in my IP camera life, thank you.

I have been switching the -o option on and off depending on use case, but to do everything I want, it looks like I'll be running 2 instances of webrtc-streamer in docker:

- "Direct" stream when I need it, with low CPU overhead, -o enabled
- "Transcoded" stream when I need a custom bitrate and size, -o disabled

This is OK, but when 2 users access the IP camera at the same time, getting streams that use both Direct and Transcoded, it causes double RTSP streams to be retrieved from the camera and doubles the network bandwidth. It also taxes the camera and I lose some frames.

I would love to have a single RTSP stream from IP camera to webrtc-streamer, and to be able to retrieve a WebRTC stream that is either transcoded, or direct, with a GET parameter. Like options=transcode=direct, which would provide a stream with the same awesome efficiency of the -o option.

Is this a possible feature? Thanks again!

mpromonet, 2021-09-06:
Hi,

It's nice this piece of code helps people to achieve other needs.

I think this is possible to achieve, but needs some adaptation, to be able to register (and then select) 2 H264 decoder, one to keep encoded frame, one that really encompress. It also need to publish into 2 different WebRTC streams, maybe implementing some tee....

Another approach may be to insert something like live555ProxyServer in the middle that run locally in order to duplicate the RTSP stream without impacting the camera ?

Best Regards,
Michel.

dpd123, 2021-09-07:
The live555 Proxy Server looks like it is an option, it seems to fit the bill for multiple subscribers to RTSP, with a single stream from the source. However it seems like RTSP -> RTSP always seems to introduce more latency with each step, so I don't know how this would impact low latency which is a requirement.

In my experiments so far, I have thought more around V4L2 devices as the relay source. First establish an RTSP -> V4L2 pipe, then v4l2:// inputs for webrtc-streamer. I can re-use the /dev/video0 pipe with tensorflow and play around with other low latency things when I'm ready for that part of the project.

I was able to use your v4l2rtspclient to get RTSP to /dev/video0
Then I can view video:///dev/video0 as a source in webrtc-streamer. Great, excellent tools!

There are some issues though, I can list a few quick pros and cons for anyone curious about this setup.

PROS:
-- the latency is effectively indistinguishable between the two cases: 
    1. RTSP->webrtc-streamer, 2. RTSP->v4l2rtspclient->webrtc-streamer
There might be a difference, but I couldn't tell, rapidly switching browser tabs. After keeping the stream open for a while, it actually seemed like case #2 had slightly less lag.

-- no difference with dropped frames or any other stream problems. I could not see any visual artifacts differentiating either case.

-- the CPU utilization seems slightly higher with case #2, but not significantly, I think in theory they should be about the same since essentially the same steps and libraries are being used. There's just a small step of piping the data through the V4l2 socket.

CONS:
-- v4l2rtspclient doesn't seem to close /dev/video* devices cleanly, about 50% of the time the v4l2 loopback kernel module must be unloaded and reloaded, or v4l2rtspclient won't reconnect the RTSP stream.

-- webrtc-streamer crashes about 1/3 of the time opening v4l2 sources, using this layout. I don't have a /dev/video* device to test with, just IP cameras, so I'm not sure if this is an issue with the loopback kernel module, it's possible the problem with v4l2rtspclient and webrtc-streamer are both related to a loopback module issue?

-- obviously, using the 2 components in a modular way, there is no single connection management. So I can't ask the streaming server for a v4l2:// connection, and have the request propagate to v4l2rtspclient who will then build the /dev/video* stream.
(This would be a fun feature to add to webrtc-streamer, to initiate v4l2rtspclient sessions, but it's probably not a common use case so I'll build it myself in node.js :)

Any ideas about the 2 crashing cases? I saw an issue filed for v4l2rtspclient with the same problem, only 1 session with a /dev/video* device before it goes stale. If you don't have any ideas about webrtc-streamer crashing from v4l2 video reqs, I'll document the problem and logs in a new issue.

Thanks for your comments!