mpromonet/webrtc-streamer #449: webrtc-streamer can't play RTSP stream from mp4 file .

teo-tsou, 2021-09-22:
**Describe the bug**
Hello!
So I created a RTSP stream from:  ffmpeg -re -stream_loop -1 -i mj.mp4 -f rtsp rtsp://localhost:8554/mystream 
My RTSP server is RTSP simple server.
I tested the RTSP link on VLC and everything looks fine.

But when I run ./webrtc-streamer rtsp://localhost:8554/mystream 

```
[000:000][4487] (audio_device_pulse_linux.cc:1605): failed to connect context, error=-1
[000:000][4487] (audio_device_pulse_linux.cc:145): failed to initialize PulseAudio
[000:000][4487] (audio_device_impl.cc:341): Audio device initialization failed
(PeerConnectionManager.h:235): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.09352491457674716
 Requested URL : rtsp://localhost:8554/mystream
```

And nothing displays. And then the connection is closed by webrtc:
`[020:918][4627] (PeerConnectionManager.cpp:768): Remove PeerConnection peerid:0.9340373556036519
`

Can you help please?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

mpromonet, 2021-09-22:
Hi Theodoros,

It seems the RTSP consumer cannot get SPS & PPS frame that are needed to uncompress H264.
Probably something in https://github.com/mpromonet/webrtc-streamer/blob/master/inc/livevideosource.h fails, you may look to verbose logs.
Looking to the code, it seems there is a problem when SPS/PPS are not in the sprops-parameters, there is no marker in such a case. I will have a look.

Best Regards,
Michel.

teo-tsou, 2021-09-22:
Is there any other way to stream a local file with webRTC?

mpromonet, 2021-09-22:
Hi,

Mkv file with H264, JPEG, VP8, VP9 frames could be streamed using `file://<path to mkv>`.

Best Regards,
Michel.

teo-tsou, 2021-09-23:
Ok problem solved. Actually, I was running webrtc streamer inside Kubernetes Pod and I couldn't reach STUN. I was able solving this by using host network.

closed 2021-09-23 by teo-tsou