mpromonet/webrtc-streamer #460: Question: can I use H264 over RTP as source?

niXman, 2021-11-20:
hi,

can I use H264 over RTP as source which translated to `localhost::9001`?
if yes - how should I specify it on the command line?
smth like this?: `webrtc-streamer -n 9001 -u rtp://localhost::9001` ?

thanks!

mpromonet, 2021-11-22:
Hi,

There was a quick implementaion following https://github.com/mpromonet/webrtc-streamer/issues/363
SDP is build from an `rtp://` url using getSdpFromRtpUrl defined in  https://github.com/mpromonet/live555helper/blob/master/inc/sdpclient.h

Best Regards,
Michel.

niXman, 2021-11-22:
thank you so much for the answer, but could you please answer my second question as well?:

how should I specify it on the command line?
smth like this?: `webrtc-streamer -n stream-9001 -u rtp://localhost:9001 -n stream-9002 -u rtp://localhost:9002` ?

thanks!

mpromonet, 2021-11-22:
Hi,

Yes, or using the json config file. This should work if webrtc-streamer is receiving an H264 video stream with payload 96 on localhost:9001.

Best Regards,
Michel.

niXman, 2021-11-22:
> This should work if webrtc-streamer is receiving an H264 video stream with payload 96 on localhost:9001.

sure, thank you!

closed 2021-11-22 by niXman