mpromonet/webrtc-streamer #465: Using H264 encoder instead of VPx

ihabZhaika, 2021-12-07:
Hey,
I am using the webcam as main video source, when I examined the video codec is it using VPx, which I assume is VP8, If I am running 2 browsers [video producer]  and Video consumer, in the Offer request I can tell it to use H264 as shown in this figure[source: https://webrtchacks.com/sdp-anatomy/]
[image]

The question how to instruct the streamer to use H264 to encode the webcam content ?

mpromonet, 2021-12-07:
Hi,

The streamer is using the WebRTC SDK, to tweak the prefered codec to encode, you need to modify the javascript side to select your prefered codec when multiple are available.

Best Regards,
Michel.

ihabZhaika, 2021-12-08:
> Hi,
> 
> The streamer is using the WebRTC SDK, to tweak the prefered codec to encode, you need to modify the javascript side to select your prefered codec when multiple are available.
> 
> Best Regards, Michel.

Hey,
I have tried but with no success[did the same for other signaling implementation and did work],
I changed the SDP value on the create offer and answer, On the console prints I can see that It is changed but the streamer did not switch to what I asked for, if needed I can upload the code snippet so you can examine it

ihabZhaika, 2021-12-20:
In the end was able to make it work.
Thanks

closed 2021-12-20 by ihabZhaika