mpromonet/webrtc-streamer #466: would you add the function to support rtmp?

zsinba, 2021-12-13:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
[image]
the system could pull the rtsp stream send to jitsi by xmpp and show it.
could you pls add the function, let the system could pull rtmp url stream and push to the jitsi?

the rtmpserver from nginx or other stream server. it does't depend on any other lib.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

mpromonet, 2021-12-13:
Hi,

I think there is no real difficulty to add rmtp support, you can add a new capturer for it in https://github.com/mpromonet/webrtc-streamer/blob/master/inc/CapturerFactory.h implementing an inherited `rtc::VideoSourceInterface<webrtc::VideoFrame>`

Feel free to submit a PR.

Best Regards,
Michel.

zsinba, 2021-12-14:
thanks so much.

mpromonet, 2022-05-13:
Hi,

This was done with https://github.com/mpromonet/webrtc-streamer/pull/496, for now only the docker image for amd64 is linked with librtmp.

Best Regards,
Michel.

zsinba, 2022-05-13:
lGreat job,