mpromonet/webrtc-streamer #467: compile error

zsinba, 2021-12-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
git clone the source code.
and build it .

`[ 89%] Building CXX object prometheus-cpp/core/CMakeFiles/core.dir/src/check_names.cc.o
[ 90%] Building CXX object prometheus-cpp/core/CMakeFiles/core.dir/src/counter.cc.o
[ 90%] Building CXX object prometheus-cpp/core/CMakeFiles/core.dir/src/detail/builder.cc.o
[ 90%] Building CXX object prometheus-cpp/core/CMakeFiles/core.dir/src/detail/ckms_quantiles.cc.o
[ 91%] Building CXX object prometheus-cpp/core/CMakeFiles/core.dir/src/detail/time_window_quantiles.cc.o
[ 91%] Building CXX object prometheus-cpp/core/CMakeFiles/core.dir/src/detail/utils.cc.o
[ 92%] Building CXX object prometheus-cpp/core/CMakeFiles/core.dir/src/family.cc.o
[ 92%] Building CXX object prometheus-cpp/core/CMakeFiles/core.dir/src/gauge.cc.o
[ 93%] Building CXX object prometheus-cpp/core/CMakeFiles/core.dir/src/histogram.cc.o
[ 93%] Building CXX object prometheus-cpp/core/CMakeFiles/core.dir/src/registry.cc.o
[ 93%] Building CXX object prometheus-cpp/core/CMakeFiles/core.dir/src/serializer.cc.o
[ 94%] Building CXX object prometheus-cpp/core/CMakeFiles/core.dir/src/summary.cc.o
[ 94%] Building CXX object prometheus-cpp/core/CMakeFiles/core.dir/src/text_serializer.cc.o
[ 95%] Linking CXX static library libprometheus-cpp-core.a
[ 95%] Built target core
Scanning dependencies of target webrtc-streamer
[ 96%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/HttpServerRequestHandler.cpp.o
[ 96%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/PeerConnectionManager.cpp.o
/root/webrtc-streamer/src/PeerConnectionManager.cpp: In member function ‘PeerConnectionManager::PeerConnectionObserver* PeerConnectionManager::CreatePeerConnection(const string&)’:
/root/webrtc-streamer/src/PeerConnectionManager.cpp:956:48: warning: ‘webrtc::SdpSemantics::kPlanB’ is deprecated [-Wdeprecated-declarations]
  956 |   config.sdp_semantics = webrtc::SdpSemantics::kPlanB;
      |                                                ^~~~~~
In file included from /root/webrtc-streamer/inc/PeerConnectionManager.h:18,
                 from /root/webrtc-streamer/src/PeerConnectionManager.cpp:24:
/root/webrtc-streamer/../webrtc/src/api/peer_connection_interface.h:173:3: note: declared here
  173 |   kPlanB [[deprecated]] = kPlanB_DEPRECATED,
      |   ^~~~~~
/root/webrtc-streamer/src/PeerConnectionManager.cpp:956:48: warning: ‘webrtc::SdpSemantics::kPlanB’ is deprecated [-Wdeprecated-declarations]
  956 |   config.sdp_semantics = webrtc::SdpSemantics::kPlanB;
      |                                                ^~~~~~
In file included from /root/webrtc-streamer/inc/PeerConnectionManager.h:18,
                 from /root/webrtc-streamer/src/PeerConnectionManager.cpp:24:
/root/webrtc-streamer/../webrtc/src/api/peer_connection_interface.h:173:3: note: declared here
  173 |   kPlanB [[deprecated]] = kPlanB_DEPRECATED,
      |   ^~~~~~
[ 97%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/desktopcapturer.cpp.o
[ 97%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/fileaudiocapturer.cpp.o
[ 97%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/filevideocapturer.cpp.o
[ 98%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/main.cpp.o
[ 98%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtpvideocapturer.cpp.o
[ 99%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtspaudiocapturer.cpp.o
[ 99%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtspvideocapturer.cpp.o
make[2]: *** No rule to make target '../webrtc/src/out/Release/obj/libwebrtc.a', needed by 'webrtc-streamer'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:177: CMakeFiles/webrtc-streamer.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
root@iZj6che1ku64ac0covmiukZ:~/webrtc-streamer#`

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] ubuntu 20.04
 - Browser [e.g. chrome, safari]  ssh , no browser.
 - Version [e.g. 22] 20.04

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

zsinba, 2021-12-15:
[image]

mpromonet, 2021-12-15:
Hi,

The error is before, it missed from libwebrtc.a lib that is build during cmake phasis.
You can re-run it running "rm -rf ../webrtc/src/out && cmake ."

Best Regards,
Michel.

closed 2021-12-20 by mpromonet