mpromonet/webrtc-streamer #469: Could not play video on iOS 15 devices

paladinlin1, 2021-12-17:
**Describe the bug**
Test on many iOS 15 devices and no one could play the video.
Tested both safari and chrome.
Windows, Ubuntu, Android is work fine.
Does anyone encounter the same problem?

**To Reproduce**
Server is use the ubuntu 18.04 and here is my command line:

docker run -p 8100:8100 -it mpromonet/webrtc-streamer -H 0.0.0.0:8100 rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov

then use the browser to check the website. In my case is http://172.16.3.310:8100
The website is work normally. But the video won't play.

**Smartphone (please complete the following information):**
 - Device: iPhone 13, iPad mini
 - OS: iOS 15
 - Browser safari, chrome
 - Version 0.6.5

**Additional context**
Chrome console log

LOGGet IceServers

LOGcreatePeerConnection  config: {"iceServers":rage:{"urls":rage:"stun:stun.l.google.com:19302"]}]}

LOGCreated RTCPeerConnnection with config: {"iceServers":rage:{"urls":rage:"stun:stun.l.google.com:19302"]}]}

LOGCreate offer:{"sdp":"v=0\r\no=- 6095527692312742485 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE 0 1 2\r\na=extmap-allow-mixed\r\na=msid-semantic: WMS\r\nm=audio 9 UDP/TLS/RTP/SAVPF 111 103 9 0 8 105 13 110 113 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:eYOU\r\na=ice-pwd:auWWJvLbQvoep5wmSjDNg1yL\r\na=ice-options:trickle\r\na=fingerprint:sha-256 4A:51:43:F5:B3:7E:F0:E6:F1:18:60:44:FE:DC:D6:60:8C:1D:EA:F0:01:DF:BE:A4:49:9A:3B:CA:F2:3E:EF:30\r\na=setup:actpass\r\na=mid:0\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=extmap:2 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:3 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:4 urn:ietf:params:rtp-hdrext:sdes:mid\r\na=extmap:5 urn:ietf:params:rtp-hdrext:sdes:rtp-stream-id\r\na=extmap:6 urn:ietf:params:rtp-hdrext:sdes:repaired-rtp-stream-id\r\na=sendrecv\r\na=msid:- ec85736f-3818-4403-9fab-7a4a0689c669\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 ISAC/16000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:105 CN/16000\r\na=rtpmap:13 CN/8000\r\na=rtpmap:110 telephone-event/48000\r\na=rtpmap:113 telephone-event/16000\r\na=rtpmap:126 telephone-event/8000\r\na=ssrc:2430564869 cname:Ak2H2uvFYEiKPDpv\r\na=ssrc:2430564869 msid:- ec85736f-3818-4403-9fab-7a4a0689c669\r\na=ssrc:2430564869 mslabel:-\r\na=ssrc:2430564869 label:ec85736f-3818-4403-9fab-7a4a0689c669\r\nm=video 9 UDP/TLS/RTP/SAVPF 96 97 98 99 100 101 102 125 104 124 106 107 108 109 127\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:eYOU\r\na=ice-pwd:auWWJvLbQvoep5wmSjDNg1yL\r\na=ice-options:trickle\r\na=fingerprint:sha-256 4A:51:43:F5:B3:7E:F0:E6:F1:18:60:44:FE:DC:D6:60:8C:1D:EA:F0:01:DF:BE:A4:49:9A:3B:CA:F2:3E:EF:30\r\na=setup:actpass\r\na=mid:1\r\na=extmap:14 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:2 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:13 urn:3gpp:video-orientation\r\na=extmap:3 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:12 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay\r\na=extmap:11 http://www.webrtc.org/experiments/rtp-hdrext/video-content-type\r\na=extmap:7 http://www.webrtc.org/experiments/rtp-hdrext/video-timing\r\na=extmap:8 http://www.webrtc.org/experiments/rtp-hdrext/color-space\r\na=extmap:4 urn:ietf:params:rtp-hdrext:sdes:mid\r\na=extmap:5 urn:ietf:params:rtp-hdrext:sdes:rtp-stream-id\r\na=extmap:6 urn:ietf:params:rtp-hdrext:sdes:repaired-rtp-stream-id\r\na=sendrecv\r\na=msid:- 749af4f4-5c07-490b-ba57-28e195559305\r\na=rtcp-mux\r\na=rtcp-rsize\r\na=rtpmap:96 H264/90000\r\na=rtcp-fb:96 goog-remb\r\na=rtcp-fb:96 transport-cc\r\na=rtcp-fb:96 ccm fir\r\na=rtcp-fb:96 nack\r\na=rtcp-fb:96 nack pli\r\na=fmtp:96 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=640c1f\r\na=rtpmap:97 rtx/90000\r\na=fmtp:97 apt=96\r\na=rtpmap:98 H264/90000\r\na=rtcp-fb:98 goog-remb\r\na=rtcp-fb:98 transport-cc\r\na=rtcp-fb:98 ccm fir\r\na=rtcp-fb:98 nack\r\na=rtcp-fb:98 nack pli\r\na=fmtp:98 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f\r\na=rtpmap:99 rtx/90000\r\na=fmtp:99 apt=98\r\na=rtpmap:100 H264/90000\r\na=rtcp-fb:100 goog-remb\r\na=rtcp-fb:100 transport-cc\r\na=rtcp-fb:100 ccm fir\r\na=rtcp-fb:100 nack\r\na=rtcp-fb:100 nack pli\r\na=fmtp:100 level-asymmetry-allowed=1;packetization-mode=0;profile-level-id=640c1f\r\na=rtpmap:101 rtx/90000\r\na=fmtp:101 apt=100\r\na=rtpmap:102 H264/90000\r\na=rtcp-fb:102 goog-remb\r\na=rtcp-fb:102 transport-cc\r\na=rtcp-fb:102 ccm fir\r\na=rtcp-fb:102 nack\r\na=rtcp-fb:102 nack pli\r\na=fmtp:102 level-asymmetry-allowed=1;packetization-mode=0;profile-level-id=42e01f\r\na=rtpmap:125 rtx/90000\r\na=fmtp:125 apt=102\r\na=rtpmap:104 VP8/90000\r\na=rtcp-fb:104 goog-remb\r\na=rtcp-fb:104 transport-cc\r\na=rtcp-fb:104 ccm fir\r\na=rtcp-fb:104 nack\r\na=rtcp-fb:104 nack pli\r\na=rtpmap:124 rtx/90000\r\na=fmtp:124 apt=104\r\na=rtpmap:106 VP9/90000\r\na=rtcp-fb:106 goog-remb\r\na=rtcp-fb:106 transport-cc\r\na=rtcp-fb:106 ccm fir\r\na=rtcp-fb:106 nack\r\na=rtcp-fb:106 nack pli\r\na=fmtp:106 profile-id=0\r\na=rtpmap:107 rtx/90000\r\na=fmtp:107 apt=106\r\na=rtpmap:108 red/90000\r\na=rtpmap:109 rtx/90000\r\na=fmtp:109 apt=108\r\na=rtpmap:127 ulpfec/90000\r\na=ssrc-group:FID 4148520593 2938439341\r\na=ssrc:4148520593 cname:Ak2H2uvFYEiKPDpv\r\na=ssrc:4148520593 msid:- 749af4f4-5c07-490b-ba57-28e195559305\r\na=ssrc:4148520593 mslabel:-\r\na=ssrc:4148520593 label:749af4f4-5c07-490b-ba57-28e195559305\r\na=ssrc:2938439341 cname:Ak2H2uvFYEiKPDpv\r\na=ssrc:2938439341 msid:- 749af4f4-5c07-490b-ba57-28e195559305\r\na=ssrc:2938439341 mslabel:-\r\na=ssrc:2938439341 label:749af4f4-5c07-490b-ba57-28e195559305\r\nm=application 9 UDP/DTLS/SCTP webrtc-datachannel\r\nc=IN IP4 0.0.0.0\r\na=ice-ufrag:eYOU\r\na=ice-pwd:auWWJvLbQvoep5wmSjDNg1yL\r\na=ice-options:trickle\r\na=fingerprint:sha-256 4A:51:43:F5:B3:7E:F0:E6:F1:18:60:44:FE:DC:D6:60:8C:1D:EA:F0:01:DF:BE:A4:49:9A:3B:CA:F2:3E:EF:30\r\na=setup:actpass\r\na=mid:2\r\na=sctp-port:5000\r\na=max-message-size:262144\r\n","type":"offer"}

LOGoffer: {"sdp":"v=0\r\no=- 7780151811604577899 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE 0 1 2\r\na=extmap-allow-mixed\r\na=msid-semantic: WMS 15961245695042183131\r\nm=audio 9 UDP/TLS/RTP/SAVPF 111 103 9 0 8 105 13 110 113 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:b+ht\r\na=ice-pwd:tomdpFU9Xx0AF2MPOh7CwJQB\r\na=ice-options:trickle\r\na=fingerprint:sha-256 F6:9B:6D:EF:E3:85:02:D9:EA:CE:A5:F4:C8:E2:56:8B:BC:97:E9:F2:5F:60:AB:95:81:D3:11:2D:02:00:36:03\r\na=setup:active\r\na=mid:0\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=extmap:2 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:3 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:4 urn:ietf:params:rtp-hdrext:sdes:mid\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 ISAC/16000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:105 CN/16000\r\na=rtpmap:13 CN/8000\r\na=rtpmap:110 telephone-event/48000\r\na=rtpmap:113 telephone-event/16000\r\na=rtpmap:126 telephone-event/8000\r\nm=video 9 UDP/TLS/RTP/SAVPF 98 99 102 125 104 124 106 107 108 109 127\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:b+ht\r\na=ice-pwd:tomdpFU9Xx0AF2MPOh7CwJQB\r\na=ice-options:trickle\r\na=fingerprint:sha-256 F6:9B:6D:EF:E3:85:02:D9:EA:CE:A5:F4:C8:E2:56:8B:BC:97:E9:F2:5F:60:AB:95:81:D3:11:2D:02:00:36:03\r\na=setup:active\r\na=mid:1\r\na=extmap:14 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:2 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:13 urn:3gpp:video-orientation\r\na=extmap:3 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:12 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay\r\na=extmap:11 http://www.webrtc.org/experiments/rtp-hdrext/video-content-type\r\na=extmap:7 http://www.webrtc.org/experiments/rtp-hdrext/video-timing\r\na=extmap:8 http://www.webrtc.org/experiments/rtp-hdrext/color-space\r\na=extmap:4 urn:ietf:params:rtp-hdrext:sdes:mid\r\na=extmap:5 urn:ietf:params:rtp-hdrext:sdes:rtp-stream-id\r\na=extmap:6 urn:ietf:params:rtp-hdrext:sdes:repaired-rtp-stream-id\r\na=sendrecv\r\na=msid:1596124 [... cut ...] :RXsf9TIIDIbj+Osx\r\nm=application 9 UDP/DTLS/SCTP webrtc-datachannel\r\nc=IN IP4 0.0.0.0\r\na=ice-ufrag:b+ht\r\na=ice-pwd:tomdpFU9Xx0AF2MPOh7CwJQB\r\na=ice-options:trickle\r\na=fingerprint:sha-256 F6:9B:6D:EF:E3:85:02:D9:EA:CE:A5:F4:C8:E2:56:8B:BC:97:E9:F2:5F:60:AB:95:81:D3:11:2D:02:00:36:03\r\na=setup:active\r\na=mid:2\r\na=sctp-port:5000\r\na=max-message-size:262144\r\n","type":"answer"}

LOGonError:call TypeError: Attempted to assign to readonly property.

LOGEnd of candidates.

LOGcodecs:[]

aminome-io, 2021-12-31:
The code 
```
e.sdp = e.sdp.split("\n").filter(function (e) {return "a=extmap-allow-mixed" !== e.trim()}).join("\n"))
```
in adapter.min.js causes this problem since RTCSessionDescription.sdp is a readonly property .
Modifying this code like
```
var sdp = e.sdp.split("\n").filter(function (e) {return "a=extmap-allow-mixed" !== e.trim()}).join("\n"))
e.sdp.replace(/.*/, sdp)
```
made a fix for me.

By the way, there is a bug in iOS 15 related to webrtc even after fixing above problem and webrtct still did not work in IOS 15 safari.
See https://bugs.webkit.org/show_bug.cgi?id=233752#c3 
That works for me and everything with webrtc-streamer is fine now.

paladinlin1, 2022-01-11:
> The code
> 
> ```
> e.sdp = e.sdp.split("\n").filter(function (e) {return "a=extmap-allow-mixed" !== e.trim()}).join("\n"))
> ```
> 
> in adapter.min.js causes this problem since RTCSessionDescription.sdp is a readonly property .
> 
> By the way, there is a bug in iOS 15 related to webrtc and webrtct still did not work in IOS 15 safari even after fixing above problem. See https://bugs.webkit.org/show_bug.cgi?id=233752#c3 That works for me and everything with webrtc-streamer is fine now.

Your code solved my problem. Thank your for your help.

mpromonet, 2022-01-11:
Hi,

Then this should be fixed upgrading adapter.js, this seems improved in commit https://github.com/webrtcHacks/adapter/commit/de0348c756b7bda11a700bf7ea9e9393cab16421#diff-853ec5d4a1af8095fc4e539fc6e2757d86f5ee638ff0a1c0c45964f223ab6bcd

Best Regards,
Michel.

closed 2022-01-15 by mpromonet