mpromonet/webrtc-streamer #471: unable to open display

mikefeng-max, 2021-12-20:
Hello, I use webrtc streamer for video streaming in centos7 environment. But I didn't succeed. As shown in the figure. Could you help me
[image]

mpromonet, 2021-12-20:
Hi,

Increasing the log level may help you to localize the problem.

Best Regards,
Michel.

mikefeng-max, 2021-12-21:
thanks mpromonet：
I am very glad to receive your reply．
The problem I encountered was that I could not create a datachannel open. Webrtc streamer cannot be used when it is placed on the ECS. However, it can be used normally on the local server.
[image]
[image]

mpromonet, 2021-12-21:
Hi,

I think opening or not the datachannel is a consequence of ICE negociation/enumeration. In the case that fails it seems candidate is not valid (172.19.45.79), it looks like an internal docker ip. 

Best Regards,
Michel.

mikefeng-max, 2021-12-23:
Hi Michel,
Thanks a lot for pointing me to the right direction. I will try it out.

closed 2021-12-23 by mikefeng-max