mpromonet/webrtc-streamer #473: Not working on RTSPS links (unifi protect Cameras)

maximoca, 2022-01-05:
**Describe the bug**
RTSPS urls give Segmentation fault: 11
I am trying to show Unifi Protect Cameras by RTSPS

**To Reproduce**
Steps to reproduce the behavior:
1. Enter any valid or invalid RTSPS url like `./webrtc-streamer rtsps://invalid.com`
2. You get  
`Version:v0.6.5/Darwin-x86_64 civetweb@v1.13 webrtc@d3eb8f1152-dirty live555helper@91b1b98
{
	"urls" : 
	{
		"rtsps://invalid.com" : 
		{
			"video" : "rtsps://invalid.com"
		}
	}
}Logger level:3
HTTP Listen at 0.0.0.0:8000
`
3. Try to open URL in browser
4. You get:
```
[000:000][259] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.5094197875949884
[000:001][259] (PeerConnectionManager.h:235): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.5094197875949884
Segmentation fault: 11
```

**Expected behavior**
Video Loads in browser

**Desktop (please complete the following information):**
 - OS: Mac OS 12
 - Browser chrome
 - Version 96

mpromonet, 2022-01-06:
Hi,

live555 is not build with rtsps support, then it will probably not work.
To get the crash origin, you can run it in a debugger.

Best Regards,
Michel.

mpromonet, 2022-05-13:
Hi,

Now it is build with rtsps support, then I close the issue, fill free to details problems if still occurs.

Best Regards,
Michel.

closed 2022-05-13 by mpromonet