mpromonet/webrtc-streamer #481: './webrtcstreamer.js' does not provide an export named 'default'

harpyeagle, 2022-01-31:
Hi its nest cam time again in the norther hemisphere - I was updating the build based on the latest docker and experienced this issue. I am using simple webrtc-streamer-element.js and the following in the html.

<webrtc-streamer url="nestcam" options="rtptransport=tcp&timeout=60"></webrtc-streamer>

**Describe the bug**
No stream displayed
Uncaught SyntaxError: The requested module './webrtcstreamer.js' does not provide an export named 'default'

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to '(https://webrtc-streamer.herokuapp.com/webrtc-streamer-element.html)'

**Expected behaviour**
Display the stream 

**Desktop (please complete the following information):**
- OS: Mac Mon
- Chrome Version 97.0.4692.99 (Official Build) (x86_64)

**Additional context**
If I use the build from last year - no issue with webrtcstreamer.js and page displays correctly

harpyeagle, 2022-01-31:
@mpromonet 
The "fix" for me for above was to change webrtc-streamer-element.js to 

import "./webrtcstreamer.js";

instead of 

import WebRtcStreamer from "./webrtcstreamer.js";

on line number 3

mpromonet, 2022-04-06:
Hi,

This is fixed in commit https://github.com/mpromonet/webrtc-streamer/commit/f19ac29f391d047b45c2e4a779da61270c06b680

Best Regards,
Michel

closed 2022-04-06 by mpromonet