mpromonet/webrtc-streamer #485: support WHIP protocol

jonassmedegaard, 2022-02-23:
This project seems a good candidate for embracing the new WebRTC-based streaming protocol WHIP.

The author of janus can introduce it much better than I can: https://www.meetecho.com/blog/whip-janus-part-ii/

mpromonet, 2022-05-13:
Hi @jonassmedegaard 

Maybe, you may submit a PR to implement it.

Best Regards,
MicheL

mpromonet, 2022-05-13:
Hi,

Thanks for the sharing, it seems quite close to the /api/call approach that receive an SDP offer and answer with an SDP answer.
Another possible interoperability is xmpp or others...

Best Regards,
Michel.

mpromonet, 2022-07-24:
Hi Jonas,

I dig a bit more in whip, and it does not seems to be interesting.
This project is, mainly, publishing webrtc stream. WHIP is for webrtc ingestion, I made some tries in  https://github.com/mpromonet/webrtc-streamer/tree/whip, I will probably merge it but for now incoming webrtc stream are not used.

Best Regards,
Michel.

closed 2022-07-24 by mpromonet

mpromonet, 2023-04-29:
Hi,

Digging a bit more in this protocol, it is not really interesting it is about injection, then I change the code to support WHEP that is about streaming.

 I made a small test with https://github.com/mpromonet/webrtc-streamer-whep that use this api through the component https://github.com/Eyevinn/whep-video-component.

Best Regards,
Michel.