mpromonet/webrtc-streamer #487: How to use Web-streamer in a simple client-server model?

iqbal-h, 2022-03-03:
Hi,
I am wondering if there is any detailed documentation on how to set up and use a web-streamer. Maybe I have missed it, please do share if there is one other than the Readme. I have installed the webstream using the readme, but I am confused about how to set up the client and server. I want to start from the basics, please bear with me. 

Let me describe a scenario and ask a few questions.

Scenario: Let's say there are two physical machines A and B. Both are running Ubuntu 18 LTS. A is a server and B is a client. Assume that both are placed on a local network and assigned private IPs, i.e., Server A IP is 192.168.1.10 and client B IP is 192.168.1.20. Server A will have a video source such as pre-recorded video or desktop-capture. B will be the one streaming and displaying the video A is sending to B.

(i) Do the client and server both require the same installation of web-streamer or client just requires a browser?

Server A:
There are two functionalities the server should support (not in parallel but one at a time): 1) streaming a locally stored .mkv video and 2) streaming the desktop capture. The questions are:
(ii) What should be the ./web-streamer command (with all required arguments) to start the server with functionality 1 and 2?
(iii) How do I attach the video source to the server? In the readme, it says URL. But how do I give the URL of the locally stored video or the desktop itself?

Client B:
It should be able to stream the video that their server A is uploading.
(iv) Will there be a URL that the client should browse to stream the video from server A?
(v) If there is also a client in web-streamer, what should be the ./web-streamer command (with all required arguments) to run the client?

(vi) Is it right to say that I do not need STUN/TURN as both A and B are in the same network?

I want to set up this simple scenario. If there is anything more that is required to set it up, then please do mention it.
Thanks.

mpromonet, 2022-03-03:
Hi,

You should be able to access to stream from MKV file using "file://" prefix (for instance file://video.mkv), and "screen://" (for instance screen://0) to access to desktop.

Best Regards,
Michel.

iqbal-h, 2022-03-04:
Hi,
Thanks. file:// worked.
But screen://0 did not work. I terminal output is:
./webrtc-streamer screen://0
Version:v0.6.5-15-gf0da88a-dirty/Linux-x86_64 civetweb@v1.15 webrtc@141007668c live555helper@91b1b98
{
	"urls" : 
	{
		"screen://0" : 
		{
			"video" : "screen://0"
		}
	}
}Logger level:3
HTTP Listen at 0.0.0.0:8000
[000:000][10239] (PeerConnectionManager.h:247): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.8474291658542249
[000:001][10239] (PeerConnectionManager.h:253): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.8474291658542249
[000:001][10238] (CapturerFactory.h:275): audiourl: idx_audioDevice:-1/0
[000:001][10242] (PeerConnectionManager.cpp:1150): Cannot create capturer video:screen://0
[000:001][10242] (PeerConnectionManager.cpp:1170): Cannot create capturer audio:screen://0
[000:002][10239] (PeerConnectionManager.h:253): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.8474291658542249
[000:024][10243] (webrtc_sdp.cc:420): Failed to parse: "". Reason: Expect line: candidate:<candidate-str>
[000:052][10243] (webrtc_sdp.cc:420): Failed to parse: "". Reason: Expect line: candidate:<candidate-str>
[000:069][10243] (webrtc_sdp.cc:420): Failed to parse: "". Reason: Expect line: candidate:<candidate-str>

when I browse 0.0.0.0:8000, it shows the following:
[image]

What am I missing?

mpromonet, 2022-03-05:
Hi,

Did you build with desktop support ?  
If gtk is not found at build time it is disabled (see https://github.com/mpromonet/webrtc-streamer/blob/master/CMakeLists.txt#L105)

Best Regards,
Michel.

iqbal-h, 2022-03-07:
Hi,

WEBRTCDESKTOPCAPTURE is ON by default.

Not sure if gtk was found at the build time. I manually installed libgtk-3-dev and libgtk-3-0 and the build the webrtc-stream. Still no luck.

The output for **cmake .** and **make** are as follows. What would be the workaound? I would really appreciate your help.

wisn@wisn:~/webrtc-streamer$ sudo cmake .
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMAKE_BUILD_TYPE = Release
WEBRTCROOT = /home/wisn/webrtc-streamer/webrtc
WEBRTCDESKTOPCAPTURE= ON
CMAKE_CXX_COMPILER_ID=GNU
-- Found Git: /usr/bin/git (found version "2.17.1") 
CMAKE_FIND_ROOT_PATH = 
ALSA_FOUND = FALSE
PulseAudio_FOUND = 0
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Performing Test CXX_SUPPORTS_STATIC_LIBGCC
-- Performing Test CXX_SUPPORTS_STATIC_LIBGCC - Success
-- Performing Test CXX_SUPPORTS_STATIC_LIBSTDCPP
-- Performing Test CXX_SUPPORTS_STATIC_LIBSTDCPP - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wisn/webrtc-streamer

wisn@wisn:~/webrtc-streamer$ sudo make
Scanning dependencies of target civetweb
[  0%] Building C object CMakeFiles/civetweb.dir/civetweb/src/civetweb.c.o
cc1: warning: command line option ‘-fno-rtti’ is valid for C++/ObjC++ but not for C
[  0%] Building CXX object CMakeFiles/civetweb.dir/civetweb/src/CivetServer.cpp.o
[  1%] Linking CXX static library libcivetweb.a
[  1%] Built target civetweb
Scanning dependencies of target liblive555helper
[  1%] Building CXX object live555helper/CMakeFiles/liblive555helper.dir/__/live/groupsock/GroupEId.cpp.o
[  1%] Building CXX object live555helper/CMakeFiles/liblive555helper.dir/__/live/groupsock/Groupsock.cpp.o
[  2%] Building CXX object live555helper/CMakeFiles/liblive555helper.dir/__/live/groupsock/GroupsockHelper.cpp.o
[  2%] Building CXX object live555helper/CMakeFiles/liblive555helper.dir/__/live/groupsock/IOHandlers.cpp.o
[  3%] Building CXX object live555helper/CMakeFiles/liblive555helper.dir/__/live/groupsock/NetAddress.cpp.o
[  3%] Building CXX object live555helper/CMakeFiles/liblive555helper.dir/__/live/groupsock/NetInterface.cpp.o
[  4%] Building C object live555helper/CMakeFiles/liblive555helper.dir/__/live/groupsock/inet.c.o
cc1: warning: command line option ‘-fno-rtti’ is valid for C++/ObjC++ but not for C
[  4%] Building CXX object live555helper/CMakeFiles/liblive555helper.dir/__/live/liveMedia/AC3AudioFileServerMediaSubsession.cpp.o
[  4%] Building CXX object live555helper/CMakeFiles/liblive555helper.dir/__/live/liveMedia/AC3AudioRTPSink.cpp.o
[... 453 lines cut ...]
   BasicNetworkManager();
   ^~~~~~~~~~~~~~~~~~~
In file included from /home/wisn/webrtc-streamer/src/main.cpp:23:0:
/home/wisn/webrtc-streamer/inc/PeerConnectionManager.h:190:16: warning: ‘virtual rtc::scoped_refptr<webrtc::PeerConnectionInterface> webrtc::PeerConnectionFactoryInterface::CreatePeerConnection(const webrtc::PeerConnectionInterface::RTCConfiguration&, std::unique_ptr<cricket::PortAllocator>, std::unique_ptr<rtc::RTCCertificateGeneratorInterface>, webrtc::PeerConnectionObserver*)’ is deprecated: Use CreatePeerConnectionOrError [-Wdeprecated-declarations]
            this);
                ^
In file included from /home/wisn/webrtc-streamer/inc/PeerConnectionManager.h:18:0,
                 from /home/wisn/webrtc-streamer/src/main.cpp:23:
/home/wisn/webrtc-streamer/webrtc/src/api/peer_connection_interface.h:1518:55: note: declared here
   virtual rtc::scoped_refptr<PeerConnectionInterface> CreatePeerConnection(
                                                       ^~~~~~~~~~~~~~~~~~~~
In file included from /home/wisn/webrtc-streamer/src/main.cpp:23:0:
/home/wisn/webrtc-streamer/inc/PeerConnectionManager.h:196:116: warning: ‘virtual rtc::scoped_refptr<webrtc::DataChannelInterface> webrtc::PeerConnectionInterface::CreateDataChannel(const string&, const webrtc::DataChannelInit*)’ is deprecated: Use CreateDataChannelOrError [-Wdeprecated-declarations]
      rtc::scoped_refptr<webrtc::DataChannelInterface>   channel = m_pc->CreateDataChannel("ServerDataChannel", NULL);
                                                                                                                    ^
In file included from /home/wisn/webrtc-streamer/inc/PeerConnectionManager.h:18:0,
                 from /home/wisn/webrtc-streamer/src/main.cpp:23:
/home/wisn/webrtc-streamer/webrtc/src/api/peer_connection_interface.h:974:52: note: declared here
   virtual rtc::scoped_refptr<DataChannelInterface> CreateDataChannel(
                                                    ^~~~~~~~~~~~~~~~~
[ 98%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtpvideocapturer.cpp.o
[ 99%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtspaudiocapturer.cpp.o
[ 99%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtspvideocapturer.cpp.o
[100%] Linking CXX executable webrtc-streamer
[100%] Built target webrtc-streamer

iqbal-h, 2022-03-07:
Also another question is that when I make changes in the webrtc/src/ and **make** again the cleans are not reflected.
I also tried **make clean** then **make** but no luck.

MY case is that I will be making several changes in the webrtc/src/ and I need to compile again and again to test it. So, how to compile new changes?

mpromonet, 2022-03-07:
Hi,

Installing gtk need to rebuild webrtc to take into account. If webrtc was built before insalling it, it is not enabled.

You can run in webrtc/src/out/Release 
`ninja webrtc rtc_json jsoncpp builtin_video_decoder_factory builtin_video_encoder_factory peerconnection p2p_server_utils task_queue default_task_queue_factory`

Best Regards,
Michel.

closed 2022-04-06 by mpromonet