mpromonet/webrtc-streamer #488: How to modify the resolution of USB camera！

luoluo888, 2022-03-07:
Because the software default USB camera is 4:3, how to change it to 16:9

mpromonet, 2022-03-09:
Hi,

You can try to set width & height of the capture, the first connection will try to set capture format. Following will use it.

Best Regards,
Michel.

luoluo888, 2022-03-10:
> Hi,
> 
> You can try to set width & height of the capture, the first connection will try to set capture format. Following will use it.
> 
> Best Regards, Michel.

Thank you. How to set the height and width of the server version

vegalou, 2022-06-26:
By my experience to re-stream lower bandwidth usage from IPCam,

You could stream out USB webcam's video to RTSP proxy by ffmpeg, specify the resolution & aspect ratio by ffmpeg options.

Call webrtc-streamer to RTSP proxy's streaming.