mpromonet/webrtc-streamer #491: No rule to make target '../webrtc/src/out/Release/obj/libwebrtc.a'

iqbal-h, 2022-03-23:
When I follow all the steps, I get the error on make where libwebrtc.a is not found.
```
~/webrtc-streamer$ sudo make
Scanning dependencies of target civetweb
[  1%] Built target civetweb
[ 85%] Built target liblive555helper
Scanning dependencies of target libv4l2cpp
[ 89%] Built target libv4l2cpp
[ 95%] Built target core
Scanning dependencies of target webrtc-streamer
make[2]: *** No rule to make target '../webrtc/src/out/Release/obj/libwebrtc.a', needed by 'webrtc-streamer'.  Stop.
CMakeFiles/Makefile2:107: recipe for target 'CMakeFiles/webrtc-streamer.dir/all' failed
make[1]: *** [CMakeFiles/webrtc-streamer.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

I have tried the following solutions but still the same error.
https://github.com/mpromonet/webrtc-streamer/issues/467
https://github.com/mpromonet/webrtc-streamer/issues/422

The strange thing is that I was able to successfully run webrtc-streamer on Ubuntu 18.04 on a PC about a month ago. But now, I am getting this error, when I am setting up webrtc-streamer on a laptop on ubuntu 18.04.

Could you please guide about the possible fix? Thanks.

mpromonet, 2022-03-23:
Hi,

libwebrtc.a is built during cmake, probably something wrong occurs.
You could try to rebuilt with somehting like `rm -rf ../webrtc/src/out && cmake . && make`.

Best Regards,
Michel.

iqbal-h, 2022-03-24:
I have tried the above along with the fresh clone of webrtc-streamer, but still the same error.

The /webrtc/src/out/Release only contains **args.gn** with following parameters:
```
is_clang=true
use_sysroot=false
rtc_include_internal_audio_device=false
rtc_include_pulse_audio=false
is_chrome_branded=true
is_debug=false
use_custom_libcxx=false
rtc_include_tests=false
rtc_enable_protobuf=false
rtc_build_examples=false
rtc_build_tools=false
treat_warnings_as_errors=false
rtc_enable_libevent=false
rtc_build_libevent=false
```

The cmake . output is as follows:
```
$ sudo cmake .
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMAKE_BUILD_TYPE = Release
WEBRTCROOT = /home/hiqbal/webrtc-streamer/../webrtc
WEBRTCDESKTOPCAPTURE= ON
CMAKE_CXX_COMPILER_ID=GNU
-- Found Git: /usr/bin/git (found version "2.17.1") 
Submodule 'civetweb' (https://github.com/civetweb/civetweb) registered for path 'civetweb'
Submodule 'getopt' (https://github.com/kimgr/getopt_port) registered for path 'getopt'
Submodule 'html' (https://github.com/mpromonet/webrtc-streamer-html) registered for path 'html'
Submodule 'libv4l2cpp' (https://github.com/mpromonet/libv4l2cpp) registered for path 'libv4l2cpp'
Submodule 'live555helper' (https://github.com/mpromonet/live555helper) registered for path 'live555helper'
Submodule 'prometheus-cpp' (https://github.com/jupp0r/prometheus-cpp) registered for path 'prometheus-cpp'
Cloning into '/home/hiqbal/webrtc-streamer/civetweb'...
Cloning into '/home/hiqbal/webrtc-streamer/getopt'...
Cloning into '/home/hiqbal/webrtc-streamer/html'...
Cloning into '/home/hiqbal/webrtc-streamer/libv4l2cpp'...
Cloning into '/home/hiqbal/webrtc-streamer/live555helper'...
Cloning into '/home/hiqbal/webrtc-streamer/prometheus-cpp'...
Submodule path 'civetweb': checked out 'eefb26f82b233268fc98577d265352720d477ba4'
Submodule path 'getopt': checked out '4173eb55a74e19e6010c7def2437aad8679112c5'
Submodule path 'html': checked out '99813c6159de9c4e26918b3df8ea14f8b5fe06e1'
Submodule path 'libv4l2cpp': checked out '2d162272eaace43c9b54787afdded03d40579e39'
Submodule path 'live555helper': checked out '2a8a52c149666e9492f448e641f483ebb4b86ca9'
Submodule path 'prometheus-cpp': checked out '4ea303fa66e4c26dc4df67045fa0edf09c2f3077'
CMAKE_FIND_ROOT_PATH = 
ALSA_FOUND = FALSE
PulseAudio_FOUND = 0
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
[... 530 lines cut ...]
x live/liveMedia/include/VP9VideoRTPSource.hh
x live/liveMedia/include/WAVAudioFileServerMediaSubsession.hh
x live/liveMedia/include/WAVAudioFileSource.hh
x live/liveMedia/include/RTSPClient.hh
x live/liveMedia/include/GenericMediaServer.hh
x live/liveMedia/include/MPEG2TransportStreamAccumulator.hh
x live/liveMedia/include/RawVideoRTPSource.hh
x live/liveMedia/include/JPEG2000VideoRTPSource.hh
x live/liveMedia/include/JPEG2000VideoRTPSink.hh
x live/liveMedia/include/MPEG2TransportStreamDemux.hh
x live/liveMedia/include/HLSSegmenter.hh
x live/liveMedia/include/TLSState.hh
x live/liveMedia/include/MIKEY.hh
x live/liveMedia/include/SRTPCryptographicContext.hh
x live/liveMedia/include/HMAC_SHA1.hh
x live/liveMedia/include/HMAC_hash.hh
OpenSSL_FOUND = FALSE
-- Performing Test CXX_SUPPORTS_STATIC_LIBGCC
-- Performing Test CXX_SUPPORTS_STATIC_LIBGCC - Success
-- Performing Test CXX_SUPPORTS_STATIC_LIBSTDCPP
-- Performing Test CXX_SUPPORTS_STATIC_LIBSTDCPP - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hiqbal/webrtc-streamer
```

iqbal-h, 2022-03-24:
By the way, how did you install cmake?
I did it with apt install. Could it be a problem?

iqbal-h, 2022-03-24:
I have found the fix. I am putting details here:

I was making a mistake of using ```sudo``` with ```cmake .```
However, there are some files in depot_tools that cmake is unable to access. So doing the following fixed this issue for me:
```
HOME$ sudo chmod 777 -R ./depot_tools
webrtc-streamer$ cmake .
```
After running the above, if some files might fail due to permission issue then run ```sudo cmake .```
Here, you can check the ```webrtc/src/our/Release``` folder. It will contain the missing libwebrtc files.

After that run ```make``` without sudo. 
The above steps fixed the issue for me. Hope it helps.

closed 2022-03-24 by iqbal-h