mpromonet/webrtc-streamer #492: CORS

jianxin123, 2022-03-29:
the console prints:
      Invalid URI. Load of media resource  failed.
      This page is in Quirks Mode. Page layout may be impacted. For Standards Mode use “<!DOCTYPE html>”
      Get IceServers
      Invalid URI. Load of media resource  failed.
      Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at http://192.168.1.122:8000/api/getIceServers. (Reason: CORS request did not succeed). Status code: (null)
      onError:getIceServers TypeError: NetworkError when attempting to fetch resource.

and I can see the frame when I use vlc ,so the rtsp source maybe has no problem.  And I use the latset Relsease version of webrtc-streamer.PLS help me thanks

jianxin123, 2022-03-29:
should I  change 192.168.1.122

mpromonet, 2022-03-29:
Hi,

If html/js file and backend api comes from same url, everything comes from same origin.
Probably you insert a proxy between webrtc-streamer and the webbrowser that remove CORS headers.

Best Regards,
Michel.

jianxin123, 2022-03-30:
sorry,its my mistake. I have solved the problem.thank you  .

closed 2022-03-30 by jianxin123