mpromonet/webrtc-streamer #493: How to specify to use public IP?

liangfj, 2022-04-01:
It only responds to the internal network IP, and there is no public network IP. How to specify the return to the public network IP?

[image]

mpromonet, 2022-04-01:
Hi,

I guess you are running inside docker ?
In such a case you could run with `--net=host` to allow STUN discovery to discover your public address.

Best Regards,
Michel.

liangfj, 2022-04-01:
Hi,
Yes,I'm running with docker, and I've used --net=host,But my network card is not configured with a public network IP,So can't get. It is currently impossible to specify an ip, right?

Best Regards,
liangfj.

mpromonet, 2022-04-01:
Hi,

You should be able to reach http server configuring your NAT, then STUN should server should detect your public ip.

Best Regards,
Michel.

liangfj, 2022-04-29:
Hi,
    In some complex network situations, network access will be forwarded through multiple layers of servers, so the obtained server ip is incorrect.

    By the way, I would like to ask, how to specify the tcp protocol to pull the video stream instead of udp. Because the network is unstable, the udp pulls the video stream will lose frames, Hope to be able to pull video stream using tcp protocol.

Best Regards,
liangfj.

mpromonet, 2022-04-29:
Hi,

To use TCP streams, tous may need to setup a Turn server instead of default stun.

Best regards
Michel

fabiano182, 2022-05-10:
Hi,
I'm facing the same problem on AWS EC2. The ICE candidates IP's are Docker Gateway IP and AWS Instance internal network IP. I'm deploying it with this command: `docker run -d --name webrtc_streamer --network=host -v $PWD/config.json:/app/config.json mpromonet/webrtc-streamer:v0.6.1 -o -H 0.0.0.0:8080 -R 50000:59999 -S0.0.0.0:3478 -s$(curl -s ifconfig.me):3478`. 

Best regards,
Fabiano

mpromonet, 2022-05-10:
Hi Fabiano,

Not sure that using internal STUN server is a good idea on a AWS EC2... why not using the default google STUN ?

Best Regards,
Michel.

fabiano182, 2022-05-11:
Hello Michel,

I'm configuring this environment in China. Almost all tests that we made indicates that users can't access cameras throught the webrtc by using google STUN. So this is the problem when using the default google STUN. The main ideia is consolidate the STUN service in a private scenario.

Since the AWS EC2 has no NAT configuration and none Firewall was set. I'm running out ideias for tests. Do you have any sugestions?

Regards,
Fabiano.

mpromonet, 2022-05-12:
Hi Fabiano,

If the firewall are openned, I don't understand why running with `-S0.0.0.0:3478 -s$(curl -s ifconfig.me):3478` doesnot work. I made some tests but it is always working even when I publish no stun server or a bad address for it. It is possible that the stun configuration is no more working (or google stun is always used even when not present in ice servers?)

Best Regards,
Michel.

Nguyen-Thanh-Tung, 2023-04-18:
@mpromonet when not present in ice servers it is only possible when you using same network with webrtc-streamer server. If you using internet you cant receive stream. I am facing the same issue. What new for fix this issue?

mpromonet, 2023-04-18:
Hi @Nguyen-Thanh-Tung,

When webbrowser and webrtc-streamer are not on same network, you need either an STUN or a TURN. By default STUN server is stun.l.google.com:19302, but you can use another one.

Best Regards,
Michel.

Nguyen-Thanh-Tung, 2023-04-18:
Thank @mpromonet 
But i have a question: I published my webrtc-streamer with ip and port public, i can telnet from other network but i can't see stream. I checked my NAT configuration, it right but only show stream when i using same network. What do you have any idea for this? Below is response when i using getIceCandidate:

> [
	{
		"candidate" : "candidate:2495648957 1 udp 2122260223 10.10.2.120 50000 typ host generation 0 ufrag gArH network-id 1 network-cost 50",
		"sdpMLineIndex" : 0,
		"sdpMid" : "0"
	}
]

Nguyen-Thanh-Tung, 2023-04-18:
I also try using Turn server but nothing happen. If you need more information, please tell me