mpromonet/webrtc-streamer #494: Binding request timed out from 192.168.0.x:62323

xingkongdeweilai, 2022-04-02:
[image]

The page cannot play videos，And non-stop Requested URL. No problem with rtsp，it can be play。

xingkongdeweilai, 2022-04-02:
I guess the problem comes from UDP，but i cannot resolve

mpromonet, 2022-04-02:
Hi,

Probably increasing log level will help you.
Anyway it is possible that your RTSP server doesnot suppot RTP over TCP, that is set by default in https://github.com/mpromonet/webrtc-streamer-html/blob/master/webrtcstreamer.json
```
options: "rtptransport=tcp&timeout=60"
```

Best Regards,
Michel.

xingkongdeweilai, 2022-04-06:
Thanks, it has been solved，the reason is that the camera encoding is set to H.264, not H.265

closed 2022-04-06 by xingkongdeweilai

smallPlanet123, 2022-12-22:
I set it to h265 and it still doesn't work

smallPlanet123, 2022-12-22:
> 

I set it to h265 and it still doesn't work