mpromonet/webrtc-streamer #495: Two similar rtsp resources, one play audio and one no play

n65785828, 2022-04-14:
I have two rtsp resources, one live video and one historical video，
the  live video （rtsp://admin:xxx@192.168.1.231:554/streaming/channels/201）  play audio
the webrtc-streamer server log show
`
Start playing sink for "video/H264" subsession

Start playing sink for "audio/PCMA" subsession
`

and the  historical video （rtsp://admin:xxxx@192.168.1.231:554/streaming/tracks/201?starttime=20220414t101520z&endtime=20220414t102020z301）no audio 

the webrtc-streamer server log only show 
`
Start playing sink for "video/H264" subsession
`

I don't know where is the problem, ? Can you give me some advice? Do I need to add any configuration?

andrewaggb, 2022-04-14:
Have you tried playing both rtsp streams in vlc media player or ffplay to confirm you get audio on the recordings?  Based on the urls I'm assuming it's a surveillance product.  Many of them don't record audio or don't record audio by default as there are different laws involved.

n65785828, 2022-04-15:
@andrewaggb  I tried it and both rtsp streams in vlc media player can play audio

n65785828, 2022-04-15:
@andrewaggb   I found that when playing the historical RTSP stream in the vlc media player, only one can be played, When  i open a new window and play the same stream, the log shows (Failed to DESCRIBE: 453 Not Enough Bandwidth), Does webrtc-streamer need to process two RTSP streams at the same time when transcoding, one for transcoding video and one for transcoding audio?

andrewaggb, 2022-04-15:
@n65785828 Thinking back I remember encountering the 453 No Enough Bandwidth one some Hikvision NVR's for some other work I'd done.  I'm pretty sure I ended up scripting it to sleep a while and try again...  I'm not an expert (or even a user) of the webrtc-streamer so there is likely somebody else who can offer more help but I'm pretty versed in RTSP.
Basically the streamer should be sending a DESCRIBE command to your camera/nvr to which it'll reply with information about codecs, bitrates, etc.  The streamer should then decide what it wants as far as video and audio goes and then issue SETUP commands to select the video and audio streams.  It seems likely there is some difference between the describe output from the live feed vs the recorded feed.  From memory I think the range: option will be different for recorded media but I suspect that's not it.  I'm wondering if the audio codec is also different.  For live you said it was 'audio/PCMA' but maybe the recorded version is using different codec, perhaps MP2L2 AAC-LC or G.726.  If you up the verbosity in vlc or ffplay it should dump the entire rtsp communication and maybe you'll find something helpful.

mpromonet, 2022-04-15:
Hi @n65785828,

Yes, the actual implementation open 2 RTSP connections (one for audio, one for video), and then SETUP one stream.

Best Regards,
Michel.

n65785828, 2022-04-18:
thanks @mpromonet ,  Regarding the problem of no audio, do I have any other solution if the video can only open one RTSP stream

mpromonet, 2022-04-18:
Hi @n65785828 

You may use a proxy like live555ProxyServer to support multiple RTSP client or update the implementation.

Best Regards,
Michel.