mpromonet/webrtc-streamer #497: cross-compile on ubuntu 18.04(x86) failed

CharlesShine, 2022-04-25:
Dear friends,

I encountered an error when cross-compiling on ubuntu(x86).I just want to compile source for aarch64 platform.
before this, I've been compiled webrtc-streamer successfully for x86 platform. when do aarch64, I was depressed.

**My process was as follows**: 
under a workdir,I got the chromium , webrtc , depot_tools and webrtc-stream
```
mkdir chromium && cd chromium
fetch --no-history --nohooks chromium
gclient sync

mkdir webrtc && cd webrtc
fetch --no-history --nohooks webrtc
gclient sync

git clone https://chromium.googlesource.com/chromium/tools/depot_tools.git

git clone https://github.com/mpromonet/webrtc-streamer.git
```
then, I followed the Dockerfile.arm64 instructions to compiled the source for aarch64. but failed.

I referenced issue https://github.com/mpromonet/webrtc-streamer/issues/338 and others, but still failed.

Besides, I found that I am executing:
```
/webrtc/src/build/linux/sysroot_scripts/install-sysroot.py --arch=arm64
```
under my webrtc folder '/webrtc/src/build/linux/', generated a arm64 sysroot diretory named '**debian_bullseye_arm64-sysroot**' was different from the CMAKE_SYSTEM_PROCESSOR(debian_sid_arm64-sysroot) set in the CMakeList.txt
I modified it in CMakeList.txt, run again, it still failed.

**Error describe**:
Error occured when ‘[100%] Linking CXX executable webrtc-streamer’，and returns as follows:
```
[100%] Linking CXX executable webrtc-streamer
/usr/lib/gcc-cross/aarch64-linux-gnu/7/../../../../aarch64-linux-gnu/bin/ld: warning: /media/aiu/CharlesShine/ShareFolder/video_stream/WebRTC/cow_webrtc_arm/webrtc/src/out/Release/obj/third_party/jsoncpp/jsoncpp/json_reader.o: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0000000
/usr/lib/gcc-cross/aarch64-linux-gnu/7/../../../../aarch64-linux-gnu/bin/ld: warning: /media/aiu/CharlesShine/ShareFolder/video_stream/WebRTC/cow_webrtc_arm/webrtc/src/out/Release/obj/third_party/jsoncpp/jsoncpp/json_value.o: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0000000
/usr/lib/gcc-cross/aarch64-linux-gnu/7/../../../../aarch64-linux-gnu/bin/ld: warning: /media/aiu/CharlesShine/ShareFolder/video_stream/WebRTC/cow_webrtc_arm/webrtc/src/out/Release/obj/third_party/jsoncpp/jsoncpp/json_writer.o: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0000000
.......
.......
../webrtc/src/out/Release/obj/libwebrtc.a(i010_buffer.o): In function `rtc::RefCountedObject<webrtc::I010Buffer>::Release() const':
i010_buffer.cc:(.text[_ZNK3rtc16RefCountedObjectIN6webrtc10I010BufferEE7ReleaseEv]+0x1c): undefined reference to `__aarch64_ldadd4_acq_rel'
collect2: error: ld returned 1 exit status
CMakeFiles/webrtc-streamer.dir/build.make:319: recipe for target 'webrtc-streamer' failed
make[2]: *** [webrtc-streamer] Error 1
CMakeFiles/Makefile2:107: recipe for target 'CMakeFiles/webrtc-streamer.dir/all' failed
make[1]: *** [CMakeFiles/webrtc-streamer.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

```
Hope someone can help me, I've been stuck here for a few days. Thanks

mpromonet, 2022-04-25:
Hi,

I clearly not try to maintain build for old system (Dockerfile.arm64 use ubuntu:22.04). However you can try to build without WebRTC use_sysroot.

Best Regards,
Michel.

CharlesShine, 2022-04-26:
Hi Michel.
I tried it yesterday based on your suggestion , but returned the following error:
```
ERROR at //build/config/sysroot.gni:64:7: Assertion failed.
      assert(
      ^-----
Missing sysroot (//build/linux/debian_bullseye_arm64-sysroot). To fix, run: build/linux/sysroot_scripts/install-sysroot.py --arch=arm64
See //build/config/sysroot.gni:65:11: 
          exec_script("//build/dir_exists.py",
          ^-----------------------------------
This is where it was set.
See //build/config/linux/pkg_config.gni:5:1: whence it was imported.
import("//build/config/sysroot.gni")
^----------------------------------
See //BUILD.gn:24:1: whence it was imported.
import("//build/config/linux/pkg_config.gni")
^-------------------------------------------
ninja: error: loading 'build.ninja': No such file or directory
RTMP_FOUND = 
OpenSSL_FOUND = FALSE
-- Configuring done
-- Generating done
-- Build files have been written to: /media/aiu/CharlesShine/ShareFolder/video_stream/WebRTC/cow_webrtc_arm/webrtc-streamer
```
cause previous error was about 'aarch64-linux-gnu', I guess it might be a compiler problem. so, I reinstalled aarch64 toolchains from source code. And then , returned the following error
```
[100%] Linking CXX executable webrtc-streamer
/media/aiu/CharlesShine/ShareFolder/video_stream/WebRTC/cow_webrtc_arm/aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/7.5.0/../../../../aarch64-linux-gnu/bin/ld: cannot find -lX11
/media/aiu/CharlesShine/ShareFolder/video_stream/WebRTC/cow_webrtc_arm/aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/7.5.0/../../../../aarch64-linux-gnu/bin/ld: cannot find -lXext
/media/aiu/CharlesShine/ShareFolder/video_stream/WebRTC/cow_webrtc_arm/aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/7.5.0/../../../../aarch64-linux-gnu/bin/ld: cannot find -lXdamage
/media/aiu/CharlesShine/ShareFolder/video_stream/WebRTC/cow_webrtc_arm/aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/7.5.0/../../../../aarch64-linux-gnu/bin/ld: cannot find -lXfixes
/media/aiu/CharlesShine/ShareFolder/video_stream/WebRTC/cow_webrtc_arm/aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/7.5.0/../../../../aarch64-linux-gnu/bin/ld: cannot find -lXcomposite
/media/aiu/CharlesShine/ShareFolder/video_stream/WebRTC/cow_webrtc_arm/aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/7.5.0/../../../../aarch64-linux-gnu/bin/ld: cannot find -lXrandr
/media/aiu/CharlesShine/ShareFolder/video_stream/WebRTC/cow_webrtc_arm/aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/7.5.0/../../../../aarch64-linux-gnu/bin/ld: cannot find -lXtst
collect2: error: ld returned 1 exit status
CMakeFiles/webrtc-streamer.dir/build.make:319: recipe for target 'webrtc-streamer' failed
make[2]: *** [webrtc-streamer] Error 1
CMakeFiles/Makefile2:107: recipe for target 'CMakeFiles/webrtc-streamer.dir/all' failed
make[1]: *** [CMakeFiles/webrtc-streamer.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```
It shows missing xorg related components，now I'm going to cross-compile and install them one by one according to xorg website (https://gitlab.freedesktop.org/xorg). Hope it works.
Anyway, thanks for such an excellent project

Best Regards,
Charles

mpromonet, 2022-04-26:
Hi Charles,

The X11 dependencies is normally detecting gtk installation, it is also possible to force running `WEBRTCDESKTOPCAPTURE` to `OFF`. Maybe you kept old build of webrtc, removing webrtc/src/out may help you ?

Best Regards,
Michel.

CharlesShine, 2022-04-28:
Hi Michel,
I've given up on compiling on Ubuntu 18.04，cause errors appear one after another.  I will try again when I have free time
But luckily I built it easily on ubuntu 22.04 yesterday.
Anyway，thank you very much for your help

Best Regards,
Charles.

closed 2022-04-28 by CharlesShine