mpromonet/webrtc-streamer #498: Use webrtc streamer to play RTSP video in Linux environment. How to install CentOS? Docker runs and cannot go on in case of error ((thread. CC: 798): unable to create pthread, error 1)

hzl1015, 2022-04-27:
**Describe the bug**
Use webrtc streamer to play RTSP video in Linux environment. How to install CentOS? Docker runs and cannot go on in case of error ((thread. CC: 798): unable to create pthread, error 1)

**To Reproduce**
1.yum install docker
2.systemctl start docker
3.docker pull mpromonet/webrtc-streamer
4.docker run -itd -p 8000:8000 --name webrtc-streamer mpromonet/webrtc-streamer
5. docker ps
6.docker logs 6b06a101826a ，  You will encounter (thread.cc:798): Unable to create pthread, error 1错误
7.telnet 127.0.0.1 8000
8.netstat -apn|grep 8000

**Expected behavior**
Want to successfully use webrtc streamer to play RTSP video in Linux environment. Using docker, you can start webrtc streamer

**Screenshots**
[image]
[image]
[image]
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version Linux version 3.10.0-1062.el7.x86_64 (mockbuild@kbuilder.bsys.centos.org) (gcc version 4.8.5 20150623 (Red Hat 4.8.5-36) (GCC) ) #1 SMP Wed Aug 7 18:08:02 UTC 2019
[image]

**Smartphone (please complete the following information):**
[image]

**Additional context**
If you want to use webrtc streamer to play RTSP video in Linux environment, you can directly run webrtc streamer and report an error. You need to upgrade glbic and GCC (abandon this method). Therefore, I used docker to install, but I also encountered an error. At present, I can't find any solution. Please ask for help

mpromonet, 2022-04-27:
Hi,

You probably need to use a kernel more up to date.

Best Regards,
Michel.

hzl1015, 2022-05-12:
OK, thank you. The problem has been solved

closed 2022-05-12 by mpromonet

lijiajun3029, 2022-07-28:
If you are worried that updating the kernel will bring additional bugs, you probably need to use an older webrtc-streamer.
for example:
`docker pull mpromonet/webrtc-streamer:v0.6.5`

caihanze, 2023-02-02:
> OK, thank you. The problem has been solved

how to solve it?