mpromonet/webrtc-streamer #502: Not supported on this platform Window X64

TotalVisit, 2022-06-02:
**Describe the bug**
Version:v0.6.5/Windows-AMD64 civetweb@v1.13 webrtc@d3eb8f1152-dirty live555helper@91b1b98
nullLogger level:3
[000:000][9568] (audio_device_generic.cc:28): BuiltInAGCIsAvailable: Not supported on this platform
[000:000][9568] (audio_device_generic.cc:38): BuiltInNSIsAvailable: Not supported on this platform

**To Reproduce**
Steps to reproduce the behavior:
run webrtc-streamer.exe

**Expected behavior**

**Screenshots**

**Desktop (please complete the following information):**
**Smartphone (please complete the following information):**

**Additional context**

mpromonet, 2022-06-02:
Hi TotalVisit,

What is the misbehavior ? 

Best Regards,
Michel.

1293662155, 2022-06-15:
i have the same problem

mpromonet, 2022-06-15:
which is ?

1293662155, 2022-06-16:
Version:v0.7.0-dirty/Windows-AMD64 civetweb@v1.15 webrtc@81d65fcf7d live555helper@d2b91cc
nullLogger level:3
[000:000][15276] (audio_device_generic.cc:28): BuiltInAGCIsAvailable: Not supported on this platform
[000:000][15276] (audio_device_generic.cc:38): BuiltInNSIsAvailable: Not supported on this platform

1293662155, 2022-06-16:
**docker also：**
Version:v0.7.0-6-g5400694/Linux-x86_64 civetweb@v1.15 webrtc@e4374a6020-dirty live555helper@d2b91cc

nullLogger level:3

[000:000][7] (adm_helpers.cc:44): Unable to set playout device.

[000:000][7] (audio_device_generic.cc:18): BuiltInAECIsAvailable: Not supported on this platform

[000:000][7] (audio_device_generic.cc:28): BuiltInAGCIsAvailable: Not supported on this platform

[000:000][7] (audio_device_generic.cc:38): BuiltInNSIsAvailable: Not supported on this platform

1293662155, 2022-06-16:
maybe i know why
i find that `BuiltInAECIsAvailable` return true When in the Android operating system

mpromonet, 2022-06-16:
Hi,

If the problem you have is the log of auto audio initialization that try different audio interface..... the process should run with  interfaces that are available. 

Best Regards,
Michel.

closed 2022-06-25 by mpromonet