mpromonet/webrtc-streamer #505: RUN in HTTPS directly

vegalou, 2022-06-14:
Try to run webrtc-streamer in HTTPS directly without R-Proxy of NGINx or Apache

Run in this way

    webrtc-streamer -C config.json -c allInOne.pem -S -s- -w ./html

Only HTTP startup, no HTTPS listening.

Or is it possible to specify SSL private key & cert configuration in config.json?

eg:

    https: {
        port: 8443,
        key: "./wild/private.key",
        cert: "./wild/fullchain.pem",
    },

Thanks.

mpromonet, 2022-06-14:
Hi,

To listen on ssl endpoint, you should add something like `-H8080r,8443s`, see https://github.com/civetweb/civetweb/blob/master/docs/UserManual.md#listening_ports-8080

Best Regards,
Michel.

vegalou, 2022-06-15:
Dear Michel:

Thanks for your reply(AGAIN), it works!! 

;-)

Here are steps to enable HTTPS for buddies.

### Create SSL PEM

Combine `private.key` & `fullchain.pem` files into one PEM file. 

	$ cat private.key fullchain.pem > all.pem

Or use text editor, append sections order like:

    -----BEGIN PRIVATE KEY-----
    -----END PRIVATE KEY-----
    -----BEGIN CERTIFICATE-----
    -----END CERTIFICATE-----
    -----BEGIN CERTIFICATE-----
    -----END CERTIFICATE-----
    -----BEGIN CERTIFICATE-----
    -----END CERTIFICATE-----

### Startup webrtc-streamer

    webrtc-streamer -C config.json -H8000r,8443s -c /PATH/all.pem -S -s- -w ./html

### Browse HTTPS

    https://YOUR.VALID.HOSTNAME:8443/webrtcstreamer.html?video=rtsp://admin:Pass1234@192.168.3.61:554/stream1

This way works webrtc-streamer streaming RTSP in WebRTC directly without R-Proxy of Web Server.

**THANKS FOR YOUR SHARE.**

@mpromonet

closed 2022-06-15 by vegalou