mpromonet/webrtc-streamer #509: Can't build v0.6.5

HaberR, 2022-06-24:
**Describe the bug**
I'm trying to run `docker build .` for v0.6.5 in order to get an image we can use. Unfortunately we can't use the v0.7.0 since it seems to be incompatible with the kernel we're using #498. When I issue the build command, it dies at the 'cmake' portion, with errors that look like:

```
/webrtc-streamer/src/HttpServerRequestHandler.cpp:304:4: note: in expansion of macro 'RTC_LOG'
  304 |    RTC_LOG(INFO) << "WS closed";
      |    ^~~~~~~
make[2]: *** [CMakeFiles/webrtc-streamer.dir/build.make:63: CMakeFiles/webrtc-streamer.dir/src/HttpServerRequestHandler.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:177: CMakeFiles/webrtc-streamer.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

**To Reproduce**
Steps to reproduce the behavior:
1. git clone the repo
2. git checkout v0.6.5
3. docker build .

**Expected behavior**
A clear and concise description of what you expected to happen.
`docker build .` should produce an image.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04.5 LTS. Saw this on 20.04.3 as well.
 - Browser [e.g. chrome, safari]: NA

**Additional context**
I know that there's a publicly accessible copy of the image, but unfortunately we can't use that since we need the `disableXframeOptions` functionality which wasn't added until later.

mpromonet, 2022-06-24:
Hi,

I will not maintain any old version, with old webrtc release, this is a free software. I just maintain latest version with latest webrtc release.
If you like to build old release you should get a compatible webrtc release.

Best Regards,
Michel.

HaberR, 2022-06-27:
Hmm, maybe I asked the wrong question. I'd actually like to be running `v0.7.0`, but when I try to run it I see:00
```
{
    "urls" : {
          ....
     }
}
Logger level:3
[000:000][1] (thread.cc:798): Unable to create pthread, error 1
```
After which the program seems to be unresponsive even to a sigint. I'm running it on Ubuntu 20.04.4 LTS. Any thoughts on what's wrong?

mpromonet, 2022-06-27:
Hi @HaberR,

You may increase log level to understand why thread cannot be created. If no thread can be created is different from exaushed the number of threads.

Best Regards,
Michel.

HaberR, 2022-06-27:
I tried executing `docker run -vvv -p 8000:8000 -it mpromonet/webrtc-streamer` but wasn't getting any more information. After some more playing around with it though, seems like the source of the issue was a stale docker version. Upgrading to anything later than `20.10.12` seems to work for us. Looks like we were using a version circa `19.03.12` before (so we're due for an upgrade...)

mpromonet, 2022-06-28:
Hi @HaberR 

Note that in order to enable log, you should give option to webrtc-streamer and not to docker run then `docker run -p 8000:8000 -it mpromonet/webrtc-streamer -vvvvv` will give more information...

Best Regards,
Michel.