mpromonet/webrtc-streamer #515: [need help] ICE connect failed.

chenkaiC4, 2022-07-08:
i want to run `webrtc-streamer ` at VM machine, which have IP: **192.168.0.173**, and the RTSP url is `rtsp://10.18.7.100:8554/api/v1/video`.  when i open the `192.168.0.173:8000` at chrome, the connection is failed.

Docker command:
```
docker run --net=host -p 8000:8000 -it mpromonet/webrtc-streamer -vvvvv rtsp://10.18.7.100:8554/api/v1/video
```

Chrome log:
[image]

Webrtc-Streamer log:
```
[009:999][16] (PeerConnectionManager.cpp:835): getIceCandidateList
[009:999][16] (HttpServerRequestHandler.cpp:82): code:200 answer:[
	{
		"candidate" : "candidate:3751426872 1 udp 2122260223 192.168.0.173 57481 typ host generation 0 ufrag 5c4k network-id 1 network-cost 50",
		"sdpMLineIndex" : 0,
		"sdpMid" : "0"
	}
]
[010:000][15] (HttpServerRequestHandler.cpp:82): code:200 answer:true
[010:000][11] (jsep_transport_controller.cc:294): Not adding candidate because the JsepTransport doesn't exist. Ignore it.
[010:038][11] (basic_port_allocator.cc:1408): Net[ens3:192.168.0.x/24:Unknown:id=1]: Allocation Phase=Relay
[010:038][11] (basic_port_allocator.cc:1546): AllocationSequence: Relay ports disabled, skipping.
[010:088][11] (basic_port_allocator.cc:1408): Net[ens3:192.168.0.x/24:Unknown:id=1]: Allocation Phase=Tcp
[010:088][11] (port.cc:190): Port[f001bb70::1:0:local:Net[ens3:192.168.0.x/24:Unknown:id=1]]: Port created with network cost 50
[010:088][11] (basic_port_allocator.cc:924): Adding allocated port for 0
[010:088][11] (basic_port_allocator.cc:946): Port[f001bb70:0:1:0:local:Net[ens3:192.168.0.x/24:Unknown:id=1]]: Added port to allocator
[010:088][11] (tcp_port.cc:179): Preparing TCP address, current state: 1
[010:088][11] (basic_port_allocator.cc:963): Port[f001bb70:0:1:0:local:Net[ens3:192.168.0.x/24:Unknown:id=1]]: Gathered candidate: Cand[:2434385864:1:tcp:1518280447:192.168.0.x:56573:local::0:5c4k:YFkmQ65/BLYpivdsfpxC5rzv:1:50:0]
[010:088][11] (basic_port_allocator.cc:996): Port[f001bb70:0:1:0:local:Net[ens3:192.168.0.x/24:Unknown:id=1]]: Port ready.
[010:088][11] (basic_ice_controller.cc:543): Sorting 0 available connections
[010:088][11] (basic_port_allocator.cc:1107): Port[f001bb70:0:1:0:local:Net[ens3:192.168.0.x/24:Unknown:id=1]]: Port completed gathering candidates.
[010:088][10] (PeerConnectionManager.cpp:1198): OnIceCandidate 0
[010:088][10] (PeerConnectionManager.cpp:1207): candidate:2434385864 1 tcp 1518280447 192.168.0.173 56573 typ host tcptype passive generation 0 ufrag 5c4k network-id 1 network-cost 50
[011:017][11] (p2p_transport_channel.cc:1328): Failed to resolve ICE candidate hostname d293606a-542a-4e48-ba4f-72eadb3e3079.local with error -2
[049:752][11] (stun_port.cc:100): Binding request timed out from 192.168.0.x:57481 (ens3)
[049:752][11] (basic_port_allocator.cc:1107): Port[f0007680:0:1:0:local:Net[ens3:192.168.0.x/24:Unknown:id=1]]: Port completed gathering candidates.
[049:752][11] (basic_port_allocator.cc:1181): All candidates gathered for 0:1:0
[049:752][11] (p2p_transport_channel.cc:1013): P2PTransportChannel: 0, component 1 gathering complete
Created new TCP socket 10 for connection
Connecting to 10.18.7.100, port 8554 on socket 10...
Created new TCP socket 10 for connection
Connecting to 10.18.7.100, port 8554 on socket 10...
Created new TCP socket 10 for connection
Connecting to 10.18.7.100, port 8554 on socket 10...
Created new TCP socket 10 for connection
Connecting to 10.18.7.100, port 8554 on socket 10...
```

TIPS:
`192.168.0.173` can ping `10.18.7.100`.

chenkaiC4, 2022-07-10:
@mpromonet Sorry to bother you, can you provide some help with my issue? Thanks.

mpromonet, 2022-07-10:
Hi @chenkaiC4 

From your log, we don't see any answer to RTSP DESCRIBE, without this it cannot work.
You may try openRTSP with your RTSP source to get more information on your problem.

Best Regards,
Michel.

closed 2025-05-17 by mpromonet