mpromonet/webrtc-streamer #516: Which version of Webrtc using for build webrtc-streamer on windows ?

tienvominh, 2022-07-13:
I'm facing a lot of errors unresolved external symbol jsoncpp.lib while complie webrtc-streamer on windows 10

  LNK2019	unresolved external symbol "public: class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl Json::Value::asString(void)const " (?asString@Value@Json@@QEBA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ) referenced in function "public: bool __cdecl RequestHandler::handle(class CivetServer *,struct mg_connection *)" (?handle@RequestHandler@@QEAA_NPEAVCivetServer@@PEAUmg_connection@@@Z)	webrtc-streamer\HttpServerRequestHandler.obj

May i did something wrong or just pick a wrong webrtc version that not matched with current webrtc-streamer.

mpromonet, 2022-07-13:
Hi,

latest webrtc doesnot build with gn configuration of this project. It was building for v0.7.0 with webrtc commit 81d65fcf7d.

Best Regards,
Michel.

tienvominh, 2022-07-14:
Hi,
I'm not sure whether i got a correct webrtc commit 81d65fcf7d. I just saw your action build C/C++ CI Windows #81: Commit [5400694](https://github.com/mpromonet/webrtc-streamer/commit/5400694d4131ae93e3d2c983818dd1bb852be08f) and tried to follow step by step but after that I get errors for command: cmake -G Ninja -DCMAKE_C_COMPILER=clang-cl.exe ...

In file included from ../../third_party/abseil-cpp\absl/utility/utility.h:51:
../../third_party/abseil-cpp\absl/meta/type_traits.h(501,3): error: static_assert failed due to requirement 'compliant || std::is_trivially_copy_assignable<std::pair<webrtc::TimeDelta, webrtc::TimeDelta>>::value': Not compliant with std::is_trivially_copy_assignable; Standard: false, Implementation: true
  static_assert(compliant || std::is_trivially_copy_assignable<T>::value,
  ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../third_party/abseil-cpp\absl/types/internal/optional.h(183,21): note: in instantiation of template class 'absl::is_trivially_copy_assignable<std::pair<webrtc::TimeDelta, webrtc::TimeDelta>>' requested here
              absl::is_trivially_copy_assignable<typename std::remove_cv<

I googed alot but not found any clue for this. Maybe there is some wrong with enviroment setting in my PC ?

mpromonet, 2022-07-14:
Hi,

Yes I told you it doesnot build with latest webrtc, the windows pipeline are failing. 

To get a specific webrtc version, you should go to ../webrtc/src and run `git checkout 81d65fcf7d`, then removing ../webrtc/src/out folder, and finally run cmake && make may work...

Best Regards,
Michel.

tienvominh, 2022-07-14:
Hi,

Just checkout 81d65fcf7d for ../webrtc/src and remove ../webrtc/src/out but the issue still there.

[134/3818] CXX obj/call/call/flexfec_receive_stream_impl.obj
FAILED: obj/call/call/flexfec_receive_stream_impl.obj
..\..\third_party\llvm-build\Release+Asserts\bin\clang-cl.exe /c ../../call/flexfec_receive_stream_impl.cc /Foobj/call/call/flexfec_receive_stream_impl.obj /nologo /showIncludes:user "-imsvcC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-imsvcC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-imsvcC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-imsvcC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\ucrt" "-imsvcC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\shared" "-imsvcC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\um" "-imsvcC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\winrt" "-imsvcC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\cppwinrt" -DUSE_AURA=1 -D_HAS_NODISCARD -D_CRT_NONSTDC_NO_WARNINGS -D_WINSOCK_DEPRECATED_NO_WARNINGS "-DCR_CLANG_REVISION=\"llvmorg-15-init-15652-g89a99ec9-1\"" -D_HAS_EXCEPTIONS=0 -D__STD_C -D_CRT_RAND_S -D_CRT_SECURE_NO_DEPRECATE -D_SCL_SECURE_NO_DEPRECATE -D_ATL_NO_OPENGL -D_WINDOWS -DCERT_CHAIN_PARA_HAS_EXTRA_FIELDS -DPSAPI_VERSION=2 -DWIN32 -D_SECURE_ATL -DWINAPI_FAMILY=WINAPI_FAMILY_DESKTOP_APP -DWIN32_LEAN_AND_MEAN -DNOMINMAX -D_UNICODE -DUNICODE -DNTDDI_VERSION=NTDDI_WIN10_FE -D_WIN32_WINNT=0x0A00 -DWINVER=0x0A00 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_INCLUDE_INTERNAL_AUDIO_DEVICE -DRTC_ENABLE_VP9 -DRTC_DAV1D_IN_INTERNAL_DECODER_FACTORY -DWEBRTC_HAVE_SCTP -DWEBRTC_USE_H264 -DWEBRTC_LIBRARY_IMPL -DWEBRTC_ENABLE_AVX2 -DRTC_ENABLE_WIN_WGC -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_WIN -DABSL_ALLOCATOR_NOTHROW=1 -D_ENABLE_EXTENDED_ALIGNED_STORAGE -DLIBYUV_DISABLE_NEON -I../.. -Igen -I../../third_party/abseil-cpp -I../../third_party/libyuv/include /W4 -Wimplicit-fallthrough -Wunreachable-code-aggressive -Wthread-safety -Wno-missing-field-initializers -Wno-unused-parameter -Wloop-analysis -Wno-unneeded-internal-declaration -Wno-nonportable-include-path -Wno-null-pointer-subtraction -Wenum-compare-conditional -Wno-psabi -Wno-ignored-pragma-optimize -Wno-unqualified-std-cast-call -Wmax-tokens -Wshadow -fno-delete-null-pointer-checks -fno-ident -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 /clang:-ffp-contract=off -fcomplete-member-pointers /Gy /FS /bigobj /utf-8 /Zc:twoPhase -ffile-reproducible /Zc:sizedDealloc- /D__WRL_ENABLE_FUNCTION_STATICS__ -fmsc-version=1916 -m64 -msse3 /Brepro -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -ffile-compilation-dir=. -no-canonical-prefixes -ftrivial-auto-var-init=pattern /O1 /Ob2 /Oy- /Zc:inline /Gw /Oi /Z7 -gcodeview-ghash -Xclang -fuse-ctor-homing /guard:cf /MT -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wexit-time-destructors -Wglobal-constructors -Wno-shadow -Wc++11-narrowing -Wundef -Wunused-lambda-capture /std:c++17 -Wno-trigraphs /TP /GR- /Fd"obj/call/call_cc.pdb"
In file included from ../../call/flexfec_receive_stream_impl.cc:11:
In file included from ../..\call/flexfec_receive_stream_impl.h:17:
In file included from ../..\call/flexfec_receive_stream.h:20:
In file included from ../..\api/rtp_headers.h:19:
In file included from ../../third_party/abseil-cpp\absl/types/optional.h:39:
In file included from ../../third_party/abseil-cpp\absl/utility/utility.h:51:
../../third_party/abseil-cpp\absl/meta/type_traits.h(501,3): error: static_assert failed due to requirement 'compliant || std::is_trivially_copy_assignable<std::pair<webrtc::FecProtectionParams, webrtc::FecProtectionParams>>::value': Not compliant with std::is_trivially_copy_assignable; Standard: false, Implementation: true
  static_assert(compliant || std::is_trivially_copy_assignable<T>::value,
  ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

I'm sorry for long log...

BRs,
Tien

mpromonet, 2022-07-14:
Hi,
After gît chechout, i forgot to tell you to run 'gclient sync', it ils need to synchronize dependencies.
Best regards
Michel

tienvominh, 2022-07-14:
Hi,
Everythings is ok now. Thank you so much!

BRs,
Tien

mpromonet, 2023-01-08:
Hi,

It is again possible to build with latest webrtc and a little patch decribed in [webrtc issue 14468](https://bugs.chromium.org/p/webrtc/issues/detail?id=14468) : 
```
 sed -i -e "s|#include <vector>|#include <vector>\n#include <cwctype>|" src/modules/desktop_capture/win/full_screen_win_application_handler.cc 
```

Best Regards,
Michel

mpromonet, 2023-01-29:
Hi @tienvominh 

As window build is working now, I am closing this issue. I integrate the fix proposed in webrtc issue in build process and will remove it when it will no more needed.

Best Regards,
Michel.

closed 2023-01-29 by mpromonet