mpromonet/webrtc-streamer #525: Question: How to use V4L2://

AllonisDave, 2022-08-15:
I have a server using four identical usb hdmi capture devices.

webrtc-streamer does find all of them and they work. however I have two problem.
USB does not maintain constant device mapping and even though these devices capture video at 1920,1080 for some reason 640,480 is always picked as the default size, and the displayed video has incorrect aspect ratio. Ideally I would like to use fixed device aliases and capture properties.

The README implies a V4L2:// uri could be used to define custom parameters but I've spent hours trying to experiment how to get this to work.

For example if a start up a GStreamer RTSP stream using the following:

./test-launch --gst-debug=0 "( alsasrc device=hw:CARD=MS2109,DEV=0 ! "audio/x-raw,channels=2,rate=48000" ! audioconvert ! opusenc ! rtpopuspay name=pay1 pt=97 v4l2src device=/dev/v4l/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.2.4.4:1.0-video-index0 ! "image/jpeg,width=720,height=480,frame-rate=20/1" ! rtpjpegpay name=pay0 pt=96 )"

and then use rtsp://127.0.0.1:8554/test in the config.json like

"Local Test" : { "video": "rtsp://192.169.0.121:8554/test" ,"audio": "rtsp://192.169.0.121:8554/test"},

I can get the desired 720,480 video and audio perfectly in sync with less than 1 sec latency.

So the question is can I simplify things by eliminating the need to create separate GStreamer RTSP streams and implement the same v4l2src and alsasrc parameters using config.json syntax like

"Local Test" : { "video": "v4l2:// ??? ", "audio": "v4l2:// ???"}

Has anyone done anything like this? Any insight will be greatly appreciated.

mpromonet, 2022-08-16:
Hi @AllonisDave,

v4l2:// url is only working with device supporting H264 capture, the purpose is to use the -o option to send encoded frames.
It extends the capturer provide by webrtc that choose an uncompressed format.   
In both case you should be able to set capture size parameter either `v4l2:///dev/video0?width=720&heigh=480` or `videocap://0?width=720&heigh=480`.   
The access by path should works with v4l2:// url.

Best Regards,
Michel.

AllonisDave, 2022-08-16:
Thanks that got me pointed in a direction.

I tried v4l2:///dev/video0?width=720&heigh=480 and it fails giving me

[NOTICE] /home/runner/work/webrtc-streamer/webrtc-streamer/libv4l2cpp/src/V4l2Device.cpp:133
	driver:uvcvideo capabilities:84a00001 mandatory:4000001
[NOTICE] /home/runner/work/webrtc-streamer/webrtc-streamer/libv4l2cpp/src/V4l2Device.cpp:136
	/dev/video0 support capture
[NOTICE] /home/runner/work/webrtc-streamer/webrtc-streamer/libv4l2cpp/src/V4l2Device.cpp:139
	/dev/video0 support streaming
[ERROR] /home/runner/work/webrtc-streamer/webrtc-streamer/libv4l2cpp/src/V4l2Device.cpp:212
	/dev/video0: Cannot set pixelformat to:H264 format is:MJPG
[ERROR] /home/runner/work/webrtc-streamer/webrtc-streamer/libv4l2cpp/src/V4l2Device.cpp:83
	Cannot init device:/dev/video0
[NOTICE] /home/runner/work/webrtc-streamer/webrtc-streamer/libv4l2cpp/src/V4l2MmapDevice.cpp:141
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
[006:695][167623] (V4l2Capturer.h:46): Failed to create V4l2Capturer(w = 0, h = 0, fps = 0)

Looking at the line
/dev/video0: Cannot set pixelformat to:H264 format is:MJPG exposes the issue.

The capabilities of the capture device looks like

/dev/video0
/dev/v4l/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.2.4.4:1.0-video-index0
video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)5/1
video/x-raw, format=(string)YUY2, width=(int)1600, height=(int)1200, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)5/1
video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)1024, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)8/1
video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)960, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)8/1
video/x-raw, format=(string)YUY2, width=(int)1360, height=(int)768, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)8/1
video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)10/1
video/x-raw, format=(string)YUY2, width=(int)1024, height=(int)768, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)10/1
video/x-raw, format=(string)YUY2, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 20/1, 10/1, 5/1 }
video/x-raw, format=(string)YUY2, width=(int)720, height=(int)576, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 25/1, 20/1, 10/1, 5/1 }
video/x-raw, format=(string)YUY2, width=(int)720, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 10/1, 5/1 }
video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 10/1, 5/1 }
image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1, 25/1, 20/1, 10/1 }
image/jpeg, width=(int)1600, height=(int)1200, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1, 25/1, 20/1, 10/1 }
image/jpeg, width=(int)1280, height=(int)1024, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1, 25/1, 20/1, 10/1 }
image/jpeg, width=(int)1280, height=(int)960, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1, 25/1, 20/1, 10/1 }
image/jpeg, width=(int)1360, height=(int)768, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1, 25/1, 20/1, 10/1 }
image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 }
image/jpeg, width=(int)1024, height=(int)768, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 }
image/jpeg, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 }
image/jpeg, width=(int)720, height=(int)576, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 }
image/jpeg, width=(int)720, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 }
image/jpeg, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 }

It's YUY2 and you need H264.

From posts in your v4l2rtspserver server project I see solutions that speak too using 

modprobe v4l2loopback video_nr=10
 v4l2compress_h264 /dev/video0 /dev/video10

I can try going down that path but first is there something that I can add to
v4l2:///dev/video0?width=720&heigh=480
that will tell it to use the image/jpeg capabilities of the capture device or is it strictly h264 only?

Thanks...

closed 2022-09-21 by mpromonet