mpromonet/webrtc-streamer #529: Syntax error in authorization file

fearlings123, 2022-09-14:
Attempting to sign onto the HTTP interface with authorization enabled results in:
`read_auth_file: syntax error in authorization file: foo=deadbeaf`

I started the service via:
` .\webrtc-streamer.exe -A .\password.txt`

Where password.txt contains:
```
foo=deadbeaf

```
I am running the Windows 0.70 AMD64 build downloaded from the Github release page.

mpromonet, 2022-09-14:
Hi,

Did you try using a password file with the format specified in https://github.com/civetweb/civetweb/blob/b5250570d5054a892e2a36859fa819525be062b9/docs/UserManual.md#global_auth_file

Best Regards;
Michel

fearlings123, 2022-09-15:
Thanks, that works. I was incorrectly using the syntax suggested in src\examples\turnserver\read_auth_file_unittest.cc

closed 2022-09-15 by fearlings123