mpromonet/webrtc-streamer #531: Open multiple video streams

hoveyLee, 2022-09-19:
**Describe the bug**
After I try to use the webpage to open multiple video streams, the vlc connection to rtsp will report an error. If the webrtc-streamer service is exited, I can connect using the vlc connection again.

**Screenshots**
[image]

mpromonet, 2022-09-19:
Hi,

Did you try to connect 2 VLC to your rtsp stream ?

Best Regards,
Michel.

hoveyLee, 2022-09-20:
[image]

Using VLC2 still fails to open，Is there a limit to the number of connections, and will it be automatically released if it exceeds?

hoveyLee, 2022-09-20:
Reading the stream with c++ opencv also fails

hoveyLee, 2022-09-20:
[image]

using the web connection service to read the stream is normal

mpromonet, 2022-09-20:
It seems your rtsp source accept only one client.

hoveyLee, 2022-09-21:
But I use VLC to connect two data sources at the same time without any problem

mpromonet, 2022-09-21:
You may look to RTSP exchanges

hoveyLee, 2022-09-21:
I'm sorry I didn't quite understand what you meant

closed 2026-10-31 by mpromonet