mpromonet/webrtc-streamer #532: Is there a way to see what bitrate is set for a feed?

iratner, 2022-09-22:
TLDR: if you don't have time for this, I get it

This is a question, and potentially a feature request.

I'm using this lib indirectly.  Our platform uses this to serve webrtc feeds.  We were initially using the front-end/js included in the repo, but I updated it to a typescript friendly async/await model.  I'm passing the API endpoints a bitrate param, but it seems to be largely ignored as the video comes through much clearer than it used to (which we don't want as it's too CPU intensive).

So my question is, are there any logs I can look at to see if the bitrate is being set correctly?  We run this lib in a docker container, and I can access its logs.  I also, obviously, have access to the API endpoints.  

Is there something in the SDP that shows this?

the `call` url is being sent like this: `http://127.0.0.1:7655/api/call?peerid=1845e710-0911-46e3-95d0-ae42de99d42f&url=high_fork_camera&options=rtptransport%253Dtcp%2526timeout%253D60%2526bitrate%253D75000%2526width%253D918%2526height%253D516`

Lastly, would requesting both audio and video vs requesting just video, change the bitrate of just the video stream?  As in, when only video is requested, all the bitrate is used for video, allowing the feed to be of higher resolution.  In the implementation[ in this repo](https://github.com/mpromonet/webrtc-streamer-html/blob/ca8fb14b1242b2242920849a89d8e814e2c577e3/webrtcstreamer.js#L18), both are requested.  In mine, I only request video.

[Here's gist](https://gist.github.com/iratner/c2c838d777a44b461c14ffbff8978072) of my code if you're curious

mpromonet, 2022-09-23:
Hi,

You can look to about://webrtc-internals to access bandwidth information.

Best Regards,
Michel.

iratner, 2022-09-23:
I guess I was hoping that api/getPeerConnectionList would return bitrate, as it returns both height and width at the moment.  Is it stored anywhere?