mpromonet/webrtc-streamer #535: Unable to play rtsp streams when started in docker mode

kalvinGit, 2022-10-12:
Unable to play rtsp streams when started in docker mode

environment:
 - OS: Linux iZwz9irdz2y12082dazq5qZ 3.10.0-1160.36.2.el7.x86_64
 - Version v0.6.5

1.docker run webrtc-streamer
`
sudo docker run -d --rm --network=host --name=wrs mpromonet/webrtc-streamer:v0.6.5 -v -H 8999 "rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0"
`
2.Unable to play rtsp
log:
```
{
        "urls" : 
        {
                "rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0" : 
                {
                        "video" : "rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0"
                }
        }
}Logger level:2
[000:000][1] (audio_device_pulse_linux.cc:1605): failed to connect context, error=-1
[000:000][1] (audio_device_pulse_linux.cc:145): failed to initialize PulseAudio
[000:000][1] (audio_device_impl.cc:341): Audio device initialization failed.
[000:000][1] (PeerConnectionManager.cpp:181): audio init fails -> disable audio capture
HTTP Listen at 8999
[031:117][12] (shared_x_display.cc:38): Unable to open display
[031:117][12] (shared_x_display.cc:38): Unable to open display
[031:417][1] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.10911564739098756
[031:419][1] (PeerConnectionManager.h:235): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.10911564739098756
[031:420][13] (audio_device_pulse_linux.cc:1605): failed to connect context, error=-1
[031:420][13] (audio_device_pulse_linux.cc:145): failed to initialize PulseAudio
Created new TCP socket 15 for connection
Connecting to 14.23.143.176, port 554 on socket 15...
[031:420][13] (audio_device_impl.cc:341): Audio device initialization failed.
[031:420][13] (CapturerFactory.h:275): audiourl: idx_audioDevice:-1/-1
[031:420][13] (PeerConnectionManager.cpp:1156): Cannot create capturer audio:rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0
[031:421][1] (media_session.cc:963): RED codec red is missing an associated payload type.
[031:422][1] (PeerConnectionManager.h:235): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.10911564739098756
[031:423][9] (usrsctp_transport.cc:652): UsrsctpTransport->OpenStream(...): Not adding data stream with sid=0 because stream is already open.
...remote connection opened
Sending request: DESCRIBE rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0 RTSP/1.0
CSeq: 2
User-Agent: LIVE555 Streaming Media v2021.11.01
Accept: application/sdp

[031:517][9] (jsep_transport_controller.cc:293): Not adding candidate because the JsepTransport doesn't exist. Ignore it.
[031:545][9] (p2p_transport_channel.cc:1324): Failed to resolve ICE candidate hostname 99c63b8f-ff69-40fd-8017-a21b634f8e92.local with error -2
[031:560][9] (jsep_transport_controller.cc:293): Not adding candidate because the JsepTransport doesn't exist. Ignore it.
Received 156 new bytes of response data.
Received a complete DESCRIBE response:
RTSP/1.0 401 Unauthorized
CSeq: 2
WWW-Authenticate: Digest realm="Login to cf0e1240aca6434e5973142f52b4c7fa", nonce="94d42d8e28e610ae3e7e2580fbb7df8a"

Resending...
Sending request: DESCRIBE rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0 RTSP/1.0
CSeq: 3
Authorization: Digest username="admin", realm="Login to cf0e1240aca6434e5973142f52b4c7fa", nonce="94d42d8e28e610ae3e7e2580fbb7df8a", uri="rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0", response="4f2e98019cef6e69048435f3bf695a5d"
User-Agent: LIVE555 Streaming Media v2021.11.01
Accept: application/sdp

[031:561][9] (jsep_transport_controller.cc:293): Not adding candidate because the JsepTransport doesn't exist. Ignore it.
[031:564][9] (jsep_transport_controller.cc:293): Not adding candidate because the JsepTransport doesn't exist. Ignore it.
Received 833 new bytes of response data.
Received a complete DESCRIBE response:
RTSP/1.0 200 OK
CSeq: 3
x-Accept-Dynamic-Rate: 1
Content-Base: rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0/
Cache-Control: must-revalidate
Content-Length: 603
Content-Type: application/sdp

v=0
o=- 2253326363 2253326363 IN IP4 0.0.0.0
s=Media Server
c=IN IP4 0.0.0.0
t=0 0
a=control:*
a=packetization-supported:DH
a=rtppayload-supported:DH
a=range:npt=now-
m=video 0 RTP/AVP 96
a=control:trackID=0
a=framerate:25.000000
a=rtpmap:96 H264/90000
a=fmtp:96 packetization-mode=1;profile-level-id=4D002A;sprop-parameter-sets=Z00AKpY1QPAET8s3BQEFQAAAAwBAAAAMoQA=,aO4xsgA=
a=recvonly
m=audio 0 RTP/AVP 97
a=control:trackID=1
a=rtpmap:97 MPEG4-GENERIC/8000
a=fmtp:97 streamtype=5;profile-level-id=1;mode=AAC-hbr;sizelength=13;indexlength=3;indexdeltalength=3;config=1588
a=recvonly

Sending request: SETUP rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0/trackID=0 RTSP/1.0
CSeq: 4
Authorization: Digest username="admin", realm="Login to cf0e1240aca6434e5973142f52b4c7fa", nonce="94d42d8e28e610ae3e7e2580fbb7df8a", uri="rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0/", response="f5662f46db14a70f59a3944bada558c8"
User-Agent: LIVE555 Streaming Media v2021.11.01
Transport: RTP/AVP/TCP;unicast;interleaved=0-1

Received 144 new bytes of response data.
Received a complete SETUP response:
RTSP/1.0 200 OK
CSeq: 4
Session: 1388536292689;timeout=60
Transport: RTP/AVP/TCP;unicast;interleaved=0-1;ssrc=3972893D
x-Dynamic-Rate: 1

 Requested URL : rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0
Start playing sink for "video/H264" subsession
markerSize:4
Sending request: SETUP rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0/trackID=1 RTSP/1.0
CSeq: 5
Authorization: Digest username="admin", realm="Login to cf0e1240aca6434e5973142f52b4c7fa", nonce="94d42d8e28e610ae3e7e2580fbb7df8a", uri="rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0/", response="f5662f46db14a70f59a3944bada558c8"
User-Agent: LIVE555 Streaming Media v2021.11.01
Transport: RTP/AVP/TCP;unicast;interleaved=2-3
Session: 1388536292689

Received a complete SETUP response:
RTSP/1.0 200 OK
CSeq: 5
Session: 1388536292689;timeout=60
Transport: RTP/AVP/TCP;unicast;interleaved=2-3;ssrc=3A48F1FE
x-Dynamic-Rate: 1

 Requested URL : rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0
Sending request: PLAY rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0/ RTSP/1.0
CSeq: 6
Authorization: Digest username="admin", realm="Login to cf0e1240aca6434e5973142f52b4c7fa", nonce="94d42d8e28e610ae3e7e2580fbb7df8a", uri="rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0/", response="781d761e74fb4a4c49fb5527e62861d9"
User-Agent: LIVE555 Streaming Media v2021.11.01
Session: 1388536292689
Range: npt=0.000-

Received a complete PLAY response:
RTSP/1.0 200 OK
CSeq: 6
Session: 1388536292689
Range: npt=0.000000-
RTP-Info: url=trackID=0;seq=29576;rtptime=31175133,url=trackID=1;seq=37836;rtptime=2802141

[046:626][72] (PeerConnectionManager.cpp:774): Remove PeerConnection peerid:0.10911564739098756
[046:626][72] (PeerConnectionManager.cpp:791): hangUp stream is no more used 9712277819617665721
[046:626][72] (PeerConnectionManager.cpp:799): hangUp stream closed 9712277819617665721
Close session: video/H264
Sending request: TEARDOWN rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0/ RTSP/1.0
CSeq: 7
Authorization: Digest username="admin", realm="Login to cf0e1240aca6434e5973142f52b4c7fa", nonce="94d42d8e28e610ae3e7e2580fbb7df8a", uri="rtsp://admin:admin123@14.23.143.176:554/cam/realmonitor?channel=1&subtype=0/", response="b92e7dc1498c33dab6336dc26f48ef1e"
User-Agent: LIVE555 Streaming Media v2021.11.01
Session: 1388536292689

[046:631][1] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.10911564739098756
[046:631][1] (PeerConnectionManager.h:235): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:5 peerid:0.10911564739098756
```

Please help me.

Thanks!

mpromonet, 2022-10-13:
Hi,

Maybe rtsp client is n'ont receiving any data. Increasing log may helps you to know this.

Best regards
Michel

kalvinGit, 2022-10-14:
> Hi,
> 
> Maybe rtsp client is not receiving any data. Increasing log may helps you to know this.
> 
> Best regards Michel

When I play a video on the web, I print a lot of this log in the background:
`
[148856:997][9] (connection.cc:1165): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=6a70615165796755412f7a33, use_candidate=0, nomination=0
`

A complete log:

```
...omit
...omit
...omit
[148856:997][9] (connection.cc:1165): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=6a70615165796755412f7a33, use_candidate=0, nomination=0
[148857:046][9] (connection.cc:1165): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=6542456c646f2f784173472f, use_candidate=0, nomination=0
[148857:095][9] (connection.cc:1165): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=497a66796842373764525535, use_candidate=0, nomination=0
[148857:143][9] (connection.cc:1165): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=374579576549394c71622f46, use_candidate=0, nomination=0
[148857:192][9] (connection.cc:1165): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=33477a413547564a50714d33, use_candidate=0, nomination=0
[148857:240][9] (connection.cc:1165): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=64575974526f676f4e79357a, use_candidate=0, nomination=0
[148857:288][9] (connection.cc:1165): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=735a7470374d575a4644736c, use_candidate=0, nomination=0
[148857:336][9] (connection.cc:1165): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=5542356e79324b2b44554833, use_candidate=0, nomination=0
[148857:385][9] (connection.cc:1165): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=5633484a3943584e56736761, use_candidate=0, nomination=0
[148857:433][9] (connection.cc:1165): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=66674b662b71727745325248, use_candidate=0, nomination=0
[148857:481][9] (connection.cc:1165): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=366478646e545661484b3562, use_candidate=0, nomination=0
[148857:529][9] (connection.cc:1165): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=33336a4b546e58506935694c, use_candidate=0, nomination=0
[148857:578][9] (connection.cc:1165): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=6c6c617a4f5a37416569714c, use_candidate=0, nomination=0
[148857:626][9] (connection.cc:1165): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=354779634c30384846307a45, use_candidate=0, nomination=0
[148857:674][9] (connection.cc:1165): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=595a664f46696f6479613353, use_candidate=0, nomination=0
[148857:722][9] (connection.cc:828): Conn[7803ec90:0:Net[eth0:172.18.112.x/20:Ethernet:id=1]:GToyXCyW:1:0:local:udp:172.18.118.x:41116->tYgzJIpB:1:1677729535:stun:udp:202.43.232.x:53613|C--I|-|0|0|7205793488602807806|-]: Timed out after 15019 ms without a response, rtt=6000
[... cut ...]
[148857:750][1] (channel.cc:522): Channel disabled: {mid: 0, media_type: audio}
[148857:750][1] (channel.cc:818): Changing voice state, recv=0 send=0 for {mid: 0, media_type: audio}
[148857:750][9] (rtp_demuxer.cc:241): Removed sink = 8f515410 bindings
[148857:750][9] (peer_connection.cc:2432): Tearing down data channel transport for mid=2
[148857:751][1] (peer_connection.cc:546): Session: 3881322336584298653 is destroyed.
[148857:751][1] (dtmf_sender.cc:240): The Dtmf provider is deleted. Clear the sending queue.
```

Can you figure out what the problem is?

Thanks!

mpromonet, 2022-10-14:
Hi,

We don't see any data coming from RTSP/RTP, there is no message "LiveVideoSource:onData", either your rtsp source doesnot support RTP over TCP or it doesnot send data.

Best Regards,
Michel.

xiaodanguoguo, 2024-01-16:
I also meet this issue, I use different machines running in docker at a same version, a docker running in a machine in North America produces the same error log, the device of rtsp source is developed in North Americam, but when I use a machine in China to connect the RTSP source from the monitor device , it can perform well althoungh a lit bit delay.  It seems existing some limitations about the network of my machine, but this log doesn't provide valuable information for us to shot the trouble. Is there any way to fix it?

xiaodanguoguo, 2024-01-16:
I have fixed it by changing the UDP port setting and eliminating the restriction of access of the UDP port, because WebRTC uses the port for UDP protocol randomly, you need to open all of the ports for clients to access.

closed 2025-01-25 by mpromonet