mpromonet/webrtc-streamer #537: Screen Capture and MKV Video File Streaming Not Working

iqbal-h, 2022-10-13:
Hi,

I have setup webrtc-streamer on ubuntu 22. The setup was successful and I have executable. 
When I run the server and open the URL on the client, the webcam streaming works fine. The browser capture works fine.
However, two cases have issue:
1. when I run server to stream MKV file. It just shows loading on the client side:
Command on server: `$./webrtc-streamer -H9090 file://home/hiqbal/s1.mkv`
The following image displays on the client side (the loading circle never ends)
[image]

2. When I try to capture screen and stream it, it only shows blank dark image:
Command on server: `$./webrtc-streamer -H9090 screen://0`
The output on the client side is as folllows:
[image]

During cmake, the WEBRTCDESKTOPCAPTURE flag is ON and GTK_Found is also 1. Below, I have pased the brief output of cmake for reference.
```
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMAKE_BUILD_TYPE = Release
WEBRTCROOT = /home/hiqbal/webrtc-streamer/../webrtc
WEBRTCDESKTOPCAPTURE= ON
CMAKE_CXX_COMPILER_ID=GNU
-- Found Git: /usr/bin/git (found version "2.34.1") 
CMAKE_FIND_ROOT_PATH = 
ALSA_FOUND = FALSE
PulseAudio_FOUND = 0
GTK3_FOUND = 1
Done. Made 1173 targets from 252 files in 212ms
[1/3799] STAMP obj/poison_software_video_codecs.stamp
[2/3799] STAMP obj/poison_rtc_json.stamp
[3/3799] STAMP obj/poison_default_task_queue.stamp
[4/3799] STAMP obj/poison_audio_codecs.stamp
[5/3799] STAMP obj/api/call_api.stamp
[6/3799] CXX obj/api/adaptation/resource_adaptation_api/resource.o
[7/3799] CXX obj/api/libjingle_peerconnection_api/data_channel_interface.o
[8/3799] CXX obj/api/libjingle_peerconnection_api/jsep.o
[9/3799] CXX obj/api/libjingle_peerconnection_api/sctp_transport_interface.o
[10/3799] CXX obj/api/media_stream_interface/media_stream_interface.o
[11/3799] CXX obj/media/rtc_media_base/media_constants.o
[12/3799] CXX obj/api/candidate/candidate.o
[13/3799] CXX obj/api/field_trials/field_trials.o
[14/3799] CXX obj/modules/video_coding/svc/scalable_video_controller/scalable_video_controller_no_layering.o
[15/3799] CXX obj/api/dtls_transport_interface/dtls_transport_interface.o
[16/3799] CXX obj/api/video/encoded_image/encoded_image.o
[17/3799] CXX obj/api/libjingle_peerconnection_api/stats_types.o
[18/3799] CXX obj/api/rtp_parameters/rtp_parameters.o
[19/3799] CXX obj/api/libjingle_peerconnection_api/rtp_receiver_interface.o
In file included from ../../api/rtp_receiver_interface.cc:11:
In file included from ../../api/rtp_receiver_interface.h:22:
In file included from ../../api/frame_transformer_interface.h:18:
In file included from ../../api/video/encoded_frame.h:19:
In file included from ../../modules/video_coding/encoded_frame.h:18:
```

I have also tried recompiling webrtc by removing webrtc/src/out and git clean webrtc-streamer. Still no luck. 
Any help would be highly appreciated.
Thanks

mpromonet, 2022-10-13:
Hi,

File streaming is only supported for file that could be demuxed by live555, looking to log may helps you to understand if your file is supported, you may try to stream it with live555MediaServer, if it works with it, it should work with webrtc-streamer.

For screen, you should try the URL generated  screen://239.

Best regards
Michel

iqbal-h, 2022-10-14:
Hi,

The URL generated screen://299 also gives the same result, i.e., black screen with seconds counter at the bottom incrementing.
Any ideas how can I fix this?

mpromonet, 2022-11-05:
Hi,

Maybe hardware accelerade rendering is not capture ?

Best Regards,
Michel.

closed 2025-01-19 by mpromonet