mpromonet/webrtc-streamer #539: Why I use WebRtcStreamer that always reconnect every 1 minute (60 sec) ?

strong0716, 2022-11-23:
Excuse me ~

Why I use WebRtcStreamer that always reconnect every 1 minute (60 sec) ? 

I use vlc to stream-out a rtsp url, and use webrtc-streamer.exe (win10) to convert the rtsp url to webrtc, then display it in html.
but that always reconnect every 1 minute in my chrome browser ?

that was my code in JS :
```
this.webRtcServer = new WebRtcStreamer("video-live", location.protocol+"//"+window.location.hostname+":8888");
that.webRtcServer.connect("rtsp://127.0.0.1:8554/");
```

and my code in CMD :

`webrtc-streamer -H 8888 -N 4 -o rtsp://127.0.0.1:8554/ -q (?=rtsp).*`

can anyone help me?

thank you very much ~~
<(_ _)>

strong0716, 2022-11-24:
Hi everybody ~
I find the problem is vlc caused.
the rtsp generated by vlc that has a session timeout limit, and the default value is 60s.
so, does webrtc-streamer.exe not send the session heart packet to rtsp server ?
this function should I do manually (in javascript) ?

mpromonet, 2022-11-24:
Hi,

The RTCP RR report are sent by live555 RTSP client. If VLC close the RTSP session, maybe some firewall is filtering it ?

Best Regards,
Michel.

strong0716, 2023-03-10:
Excuse me..

recently, I use webrtc-streamer.exe to connect a AXIS Camera not VLC rtsp server,
the problem is occur again...

in VLC server, I can adjust the server config to set RTSP session timeout to zero, and that always keep connecting
but now, the AXIS camera can not set this config in the rtsp server whitch build in camera hardware

How can I fix this problem ?

(ps.) I was tried to use this webrtc-streamer to connect another github project called "rtsp-simple-server", and it was also have this problem

thank you very much  <(_ _)>

strong0716, 2023-03-10:
oh ....
I look the similar issue before
and I tried to connect the rtsp server by "rtp on tcp" 
that was useful, it was not reconnected

thank you very much  <(_ _)>

closed 2025-05-17 by mpromonet