mpromonet/webrtc-streamer #552: if I use param -o, it usually cannot show video

libofei2004, 2023-01-24:
I use webrtc-streamer in lan and with a public ip in wan, the cameras are in lan, if I use cmd like: webrtc-streamer.exe -H 192.168.0.102:8000, I can visit videos in internet successfully, but if i use cmd with -o, like: webrtc-streamer.exe -H 192.168.0.102:8000 -o, I cannot visit videos in internet and the cmd reports:

Unable to determine our source address: This computer does not have a valid IP (v4 or v6) address!
Unable to determine our source address: This computer does not have a valid IP (v4 or v6) address!
 Requested URL : rtsp://admin:gsj12345@192.168.0.249:554/h264/ch01/main/av_stream
Start playing sink for "video/H264" subsession
markerSize:4
Unable to determine our source address: This computer does not have a valid IP (v4 or v6) address!
Unable to determine our source address: This computer does not have a valid IP (v4 or v6) address!
 Requested URL : rtsp://admin:gsj12345@192.168.0.249:554/h264/ch01/main/av_stream
[127:816][11144] (PeerConnectionManager.h:251): OnRenegotiationNeeded peerid:0.15099549486110275
[127:821][11144] (PeerConnectionManager.h:257): OnSignalingChange state:3 peerid:0.15099549486110275
[127:824][11144] (PeerConnectionManager.h:257): OnSignalingChange state:0 peerid:0.15099549486110275
[127:836][11144] (PeerConnectionManager.h:251): OnRenegotiationNeeded peerid:0.5098032805847752
[127:840][11144] (PeerConnectionManager.h:257): OnSignalingChange state:3 peerid:0.5098032805847752
[127:844][11144] (PeerConnectionManager.h:257): OnSignalingChange state:0 peerid:0.5098032805847752
[128:394][11144] (PeerConnectionManager.h:160): OnStateChange channel:ServerDataChannel state:open
[128:412][11144] (PeerConnectionManager.h:247): OnDataChannel
[128:412][11144] (PeerConnectionManager.h:160): OnStateChange channel:ClientDataChannel state:open
[128:412][11144] (PeerConnectionManager.h:167): OnMessage channel:ClientDataChannel msg:local channel openned
[128:417][11144] (PeerConnectionManager.h:167): OnMessage channel:ServerDataChannel msg:remote channel openned
[128:430][11144] (PeerConnectionManager.h:160): OnStateChange channel:ServerDataChannel state:open
[128:470][11144] (PeerConnectionManager.h:247): OnDataChannel
[128:471][11144] (PeerConnectionManager.h:160): OnStateChange channel:ClientDataChannel state:open
[128:471][11144] (PeerConnectionManager.h:167): OnMessage channel:ClientDataChannel msg:local channel openned
[128:471][11144] (PeerConnectionManager.h:167): OnMessage channel:ServerDataChannel msg:remote channel openned
[164:589][12320] (stun_port.cc:307): Port[816f23e0:0:1:0:local:Net[Realtek:192.168.0.x/32:Wifi:id=2]]: UDP send of 1207 bytes to host 123.113.47.x:57905 (123.113.47.x:57905) failed with error 10035
[164:589][12320] (stun_port.cc:307): Port[816f23e0:0:1:0:local:Net[Realtek:192.168.0.x/32:Wifi:id=2]]: UDP send of 1207 bytes to host 123.113.47.x:57905 (123.113.47.x:57905) failed with error 10035

yangjieshao, 2023-03-08:
海康现在的设备默认的流都是H265的 去设备管理页面把主码流改成H264试试

libofei2004, 2023-03-08:
> 海康现在的设备默认的流都是H265的 去设备管理页面把主码流改成H264试试

@yangjieshao 谢谢回复，这个跟-o参数没什么关系，如果不调成264，就算没有-o也根本出不来，能播出来的前提肯定得先用H264

yangjieshao, 2023-03-08:
@libofei2004 
你是在公网看不见内网视频？
- -你好像 没配置turn服务器

libofei2004, 2023-03-08:
@yangjieshao 不用-o参数可以播，用了就播不了

yangjieshao, 2023-03-08:
> @yangjieshao 嗯，turn我也配过。我现在主要试验不用turn的情况。你外网用的是动态域名映射服务吗？
@libofei2004 
我turn服务器是部署在腾讯云.
然后 因为公司的路由器有公网IP 所以就用了ddns.
另一个项目上是用frp把内网的 8000 端口映射到公网服务器使用，效果一样.
但是要注意，如果内网有防火墙拦截，所有的流量会走turn服务做转发，没法走P2P，对turn服务器的带宽压力很大.
没防火墙的话，视频的流量是P2P的，对turn服务器压力就很小了

libofei2004, 2023-03-08:
@yangjieshao 没有防火墙的情况下，如果不用turn服务转发，你试过加-o参数能播到外网吗

libofei2004, 2023-03-08:
@yangjieshao 用STUN是可以做穿透的，turn应该只是中转。谷歌的STUN服务器在内地也能访问，就算不能访问，也可以使用其他的服务器

yangjieshao, 2023-03-08:
> @yangjieshao 用STUN是可以做穿透的，turn应该只是中转。谷歌的STUN服务器在内地也能访问，就算不能访问，也可以使用其他的服务器

@libofei2004   
谷歌的stun 反正我不开vpn是访问不到的
关掉它吧
只用turn服务就可以外网p2p访问内网的设备了

至于为什么不用别的stun服务器  - - 单纯只是因为我找不到可用安装包

yangjieshao, 2023-03-08:
@libofei2004 
[image]
https://github.com/mpromonet/webrtc-streamer/issues/493#issuecomment-1113150816

libofei2004, 2023-03-09:
@yangjieshao 这么看也有可能

closed 2024-05-18 by mpromonet