mpromonet/webrtc-streamer #554: Not building docker image

sun-path, 2023-01-27:
**Describe the bug** 
Building the docker image it fails with a failed to download fonts file. (test_fonts.tar.gz)

**To Reproduce**
Try to build the docker image.

**Expected behavior**
I expected the code to compile and if fails to download the test_fonts.tar.gz from google storage.

**Screenshots**
`________ running 'download_from_google_storage --no_resume --extract --no_auth --bucket chromium-fonts -s src/third_party/test_fonts/test_fonts.tar.gz.sha1' in '/webrtc'
Failed to fetch file gs://chromium-fonts/336e775eec536b2d785cc80eff6ac39051931286 for src/third_party/test_fonts/test_fonts.tar.gz. [Err: ]
0> Downloading src/third_party/test_fonts/test_fonts.tar.gz@336e775eec536b2d785cc80eff6ac39051931286...
0> Failed to fetch file gs://chromium-fonts/336e775eec536b2d785cc80eff6ac39051931286 for src/third_party/test_fonts/test_fonts.tar.gz, skipping. [Err: ]
Downloading 1 files took 1.367625 second(s)
Error: Command 'download_from_google_storage --no_resume --extract --no_auth --bucket chromium-fonts -s src/third_party/test_fonts/test_fonts.tar.gz.sha1' returned non-zero exit status 1 in /webrtc
Failed to fetch file gs://chromium-fonts/336e775eec536b2d785cc80eff6ac39051931286 for src/third_party/test_fonts/test_fonts.tar.gz. [Err: ]
0> Downloading src/third_party/test_fonts/test_fonts.tar.gz@336e775eec536b2d785cc80eff6ac39051931286...
0> Failed to fetch file gs://chromium-fonts/336e775eec536b2d785cc80eff6ac39051931286 for src/third_party/test_fonts/test_fonts.tar.gz, skipping. [Err: ]
Downloading 1 files took 1.367625 second(s)
The command '/bin/sh -c apt-get update && DEBIAN_FRONTEND=noninteractive apt-get install -y --no-install-recommends ca-certificates wget git python3 python3-pkg-resources g++ autoconf automake libtool xz-utils libpulse-dev libasound2-dev libgtk-3-dev libxtst-dev libssl-dev librtmp-dev cmake make pkg-config p7zip-full 	&& git clone https://chromium.googlesource.com/chromium/tools/depot_tools.git /depot_tools 	&& export PATH=/depot_tools:$PATH 	&& mkdir /webrtc 	&& cd /webrtc 	&& fetch --no-history --nohooks webrtc 	&& sed -i -e "s|'src/resources'],|'src/resources'],'condition':'rtc_include_tests==true',|" src/DEPS 	&& gclient sync 	&& cd /webrtc-streamer 	&& cmake . && make 	&& cpack 	&& mkdir /app && tar xvzf webrtc-streamer*.tar.gz --strip=1 -C /app/ 	&& rm -rf /webrtc && rm -f *.a && rm -f src/*.o 	&& apt-get clean && rm -rf /var/lib/apt/lists/' returned a non-zero code: 2
Cleaning up project directory and file based variables 00:01
ERROR: Job failed: exit code 2`

**Desktop (please complete the following information):**
I tried this on a AMD64 ubuntu box and a Gitlab pipeline. 

**Additional context**
Would appreciate if you could run your pipelines again just so we know for sure it's on my end or not.

mpromonet, 2023-01-28:
Hi,

I rerun the pipeline and all failed, for different reason, local build worked, but It was not a clean build.
Need to investigate.

Best Regards,
Michel.

sun-path, 2023-01-31:
OK. I'll try that and update here later on today.

sun-path, 2023-02-03:
Builds fine now! Thanks for your assistance.

closed 2023-02-03 by sun-path