mpromonet/webrtc-streamer #555: [NPM] Expose underlying mediastreams

LeoDog896, 2023-01-31:
**Is your feature request related to a problem? Please describe.**
I have agonistic camera displays that I would like to use that take in a MediaStream. However, webrtc-streamer (js) doesn't expose the underlying mediastream objects for my use

**Describe the solution you'd like**
I would like an exposed array of the media streams for my own display purposes.

**Describe alternatives you've considered**
- Not using the agonistic display (but for the sake of my own team & mocking displays in codespaces, this is not preferred)

mpromonet, 2023-02-01:
Hi,

The html files are a separate project, just a simple implementation. Using api /api/call you could create webrtc peerconnection.

Best regards
Michel

LeoDog896, 2023-02-01:
Thanks! My team is mainly a web development team so we were under the impression that the webrtc-streamer NPM package was first-class, not an example package. Thanks for the clarification!

closed 2023-02-01 by LeoDog896