mpromonet/webrtc-streamer #558: The CPU occupation of instruction v0.7.1 is significantly higher than that of instruction v0.7.0 by command -o

yangjieshao, 2023-02-04:
在v0.7.1 使用-o命令
Use the - o command in v0.7.1
出现大量的 "The ROI position protrudes beyond the right edge of the image, Ignore roi_x"日志
Too many of logs  "The ROI position protrudes beyond the right edge of the image, Ignore roi_x"
CPU占用明显提升
Significant increase in CPU consumption

Video： H.264  1920*1080 25fps  2048 Kbps (upper limit)
**Screenshots**
v0.7.1
[image]

v0.7.0
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: PC
 - OS:  Windows 10 22H2 19045.2251
          J900  4G RAM
 [image]

 - Browser Chrome 109.0.5414.120
 [image]

 - Version v 0.7.1
[image]

mpromonet, 2023-02-24:
Hi
Using roi parameters with -o are incompatible. Without decoding cropping is probably impossible.
Best Regards
Michel

yangjieshao, 2023-02-25:
> Hi Using roi parameters with -o are incompatible. Without decoding cropping is probably impossible. Best Regards Michel

我明白你的意思，但是在我的问题中，我并没有想要使用roi进行裁剪
(I understand what you mean, but in my question, I don't want to use roi )

我使用的启动命令如下:
(The startup command I use is as follows)
-a -o -s- -Tname:password@127.0.0.1:3478 -H 0.0.0.0:8000 -R 20000:65535

同样的启动命令，在同一台电脑，播放同一个rtsp
(The same startup command plays the same rtsp on the same computer)

在 Version v 0.7.0 或更早的版本前没有问题
(No problem before Version v 0.7.0 or earlier)

但是在 Version v 0.7.1 忽然出现了大量的日志提示ROI忽略
(However, in Version v 0.7.1, a large number of logs prompt that ROI is ignored)
[image]

同一路视频会持续不间断的写同样的roi被忽略的信息
(The same video will continuously write the same roi ignored information)
我认为这个日志最多在启动这一路视频的时候打印一次即可，过多的打印会淹没有效日志
(I think this log can be printed at most once when the video is started. Too much printing will drown the log)

在 Version v 0.7.0 或更早的版本，播放一路视频，CPU占用基本为0
(In Version v 0.7.0 or earlier, when playing a video, the CPU consumption is basically 0)
[image]
但是在 Version v 0.7.1 播放一路视频CPU占用达到了2.7%左右
(However, in Version v 0.7.1, the CPU consumption of playing one way of video reaches about 2.7%)
[image]

可能是因为我的CPU是J1900,性能比较差, 所以在大量打印ROI被忽略的日志的时候CPU占用明显提升
(It may be because my CPU is J1900 and the performance is poor, so the CPU consumption is significantly improved when printing a large number of logs with ROI ignored)
但是在之前的版本J1900完全胜任需求
(However, the previous version of J1900 is fully qualified)
由于日志的大量打印要提升CPU性能是否并不太合理
(Is it not reasonable to improve CPU performance due to large amount of log printing)

mpromonet, 2023-03-06:
Hi,

I cannot reproduce your problem, but error checking roi_x may be printed when frame.width() is 0. So I made small modification and add mors logs for further troublesooting.

Best Regards,
Michel.

yangjieshao, 2023-03-08:
> Hi,
> 
> I cannot reproduce your problem, but error checking roi_x may be printed when frame.width() is 0. So I made small modification and add mors logs for further troublesooting.
> 
> Best Regards, Michel.

我注意到您的新的修改会每一帧都打印一个 LS_VERBOSE 日志
(I noticed that your new modification will print a LS_VERBOSE log for each frame)
我的问题恰恰是在有限的低性能CPU上频繁打印日志导致CPU占用产生了预期外的占用
(My problem is precisely that frequent log printing on limited low performance CPUs results in unexpected CPU usage)
使用什么启动命令可以不答应  LS_VERBOSE 日志？
(So what start command can be used to disallow LS_ VERBOSE log?)

mpromonet, 2023-03-08:
Hi
By default only level >= info are printed.
Best Regards
Michel

closed 2023-03-11 by yangjieshao