mpromonet/webrtc-streamer #560: can not see video with -o  in janus website

GuoSirius, 2023-02-15:
execute command:
`docker run -idt --network=host --name webrtc-streamer mpromonet/webrtc-streamer -o -vvvvv -a0 -N100 -H 0.0.0.0:18888 -R 0:65535 -T0.0.0.0:3479 -tturn:turn@$(curl -s ifconfig.me):3479 -S0.0.0.0:3478 -s$(curl -s ifconfig.me):3478 -C /app/config.json`

with -o  janus join fail : `onSetAnswer:{"error":"Can't get remote description."}`

without -o janus join success

jianxin123, 2023-02-21:
i use -o also can not see the video in firefox . And the video shows without "-o",but the cpu is too high
@mpromonet

mpromonet, 2023-02-21:
Hi,

The -o option is supported only with H264 backend stream, and browser need to support H264 stream coming from backend. Using NullDecoder/NullEncoder doesnot adapt stream.

Best Regards,
Michel.

jianxin123, 2023-02-22:
em... I use chrome today,it works .   But, i used firefox before ,it also worked and fails now.