mpromonet/webrtc-streamer #565: windows build issue undefined symbol: GetDpiForMonitor

mo5245, 2023-03-08:
Issue with windows build, followed the docker build file, had initial issues that it wanted 10.0.22621.0 and wasn't found added
`choco install windows-sdk-11-version-22h2-all`

Getting undefined symbol: GetDpiForMonitor running ninja, seems something else is missing or incompatible with my build environment?

```
cmd.exe /C "cd . && "C:\Program Files\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\webrtc-streamer.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100203~1.0\x64\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100203~1.0\x64\mt.exe --manifests  -- C:\webrtc\src\third_party\llvm-build\Release+Asserts\bin\lld-link.exe /nologo C:\webrtc\src\out\Release\obj\third_party\jsoncpp\jsoncpp\json_reader.obj C:\webrtc\src\out\Release\obj\third_party\jsoncpp\jsoncpp\json_value.obj C:\webrtc\src\out\Release\obj\third_party\jsoncpp\jsoncpp\json_writer.obj C:\webrtc\src\out\Release\obj\p2p\p2p_server_utils\stun_server.obj C:\webrtc\src\out\Release\obj\p2p\p2p_server_utils\turn_server.obj CMakeFiles\webrtc-streamer.dir\src\HttpServerRequestHandler.cpp.obj CMakeFiles\webrtc-streamer.dir\src\PeerConnectionManager.cpp.obj CMakeFiles\webrtc-streamer.dir\src\desktopcapturer.cpp.obj CMakeFiles\webrtc-streamer.dir\src\fileaudiocapturer.cpp.obj CMakeFiles\webrtc-streamer.dir\src\filevideocapturer.cpp.obj CMakeFiles\webrtc-streamer.dir\src\main.cpp.obj CMakeFiles\webrtc-streamer.dir\src\rtpvideocapturer.cpp.obj CMakeFiles\webrtc-streamer.dir\src\rtspaudiocapturer.cpp.obj CMakeFiles\webrtc-streamer.dir\src\rtspvideocapturer.cpp.obj  /out:webrtc-streamer.exe /implib:webrtc-streamer.lib /pdb:webrtc-streamer.pdb /version:0.0 /machine:x64 /INCREMENTAL:NO /subsystem:console  ..\webrtc\src\out\Release\obj\webrtc.lib  ..\webrtc\src\out\Release\obj\rtc_base\rtc_json.lib  civetweb.lib  getopt.lib  secur32.lib  dmoguids.lib  wmcodecdspuuid.lib  strmiids.lib  msdmo.lib  winmm.lib  dxgi.lib  d3d11.lib  iphlpapi.lib  dwmapi.lib  prometheus-cpp\core\prometheus-cpp-core.lib  live555helper\liblive555helper.lib  ..\webrtc\src\out\Release\obj\third_party\boringssl\boringssl.lib  ws2_32.lib  kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib && cd ."
LINK: command "C:\webrtc\src\third_party\llvm-build\Release+Asserts\bin\lld-link.exe /nologo C:\webrtc\src\out\Release\obj\third_party\jsoncpp\jsoncpp\json_reader.obj C:\webrtc\src\out\Release\obj\third_party\jsoncpp\jsoncpp\json_value.obj C:\webrtc\src\out\Release\obj\third_party\jsoncpp\jsoncpp\json_writer.obj C:\webrtc\src\out\Release\obj\p2p\p2p_server_utils\stun_server.obj C:\webrtc\src\out\Release\obj\p2p\p2p_server_utils\turn_server.obj CMakeFiles\webrtc-streamer.dir\src\HttpServerRequestHandler.cpp.obj CMakeFiles\webrtc-streamer.dir\src\PeerConnectionManager.cpp.obj CMakeFiles\webrtc-streamer.dir\src\desktopcapturer.cpp.obj CMakeFiles\webrtc-streamer.dir\src\fileaudiocapturer.cpp.obj CMakeFiles\webrtc-streamer.dir\src\filevideocapturer.cpp.obj CMakeFiles\webrtc-streamer.dir\src\main.cpp.obj CMakeFiles\webrtc-streamer.dir\src\rtpvideocapturer.cpp.obj CMakeFiles\webrtc-streamer.dir\src\rtspaudiocapturer.cpp.obj CMakeFiles\webrtc-streamer.dir\src\rtspvideocapturer.cpp.obj /out:webrtc-streamer.exe /implib:webrtc-streamer.lib /pdb:webrtc-streamer.pdb /version:0.0 /machine:x64 /INCREMENTAL:NO /subsystem:console ..\webrtc\src\out\Release\obj\webrtc.lib ..\webrtc\src\out\Release\obj\rtc_base\rtc_json.lib civetweb.lib getopt.lib secur32.lib dmoguids.lib wmcodecdspuuid.lib strmiids.lib msdmo.lib winmm.lib dxgi.lib d3d11.lib iphlpapi.lib dwmapi.lib prometheus-cpp\core\prometheus-cpp-core.lib live555helper\liblive555helper.lib ..\webrtc\src\out\Release\obj\third_party\boringssl\boringssl.lib ws2_32.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:webrtc-streamer.exe.manifest" failed (exit code 1) with the following output:
lld-link: error: undefined symbol: GetDpiForMonitor
>>> referenced by .\..\..\modules\desktop_capture\win\screen_capture_utils.cc:152
>>>               webrtc.lib(screen_capture_utils.obj):(class webrtc::DesktopVector __cdecl webrtc::GetDpiForMonitor(struct HMONITOR__*))
ninja: build stopped: subcommand failed.
```

mpromonet, 2023-03-12:
Hi @mo5245 

A dll was missing, I fixed this in commit https://github.com/mpromonet/webrtc-streamer/commit/4e05abbb4c41f573edb8116d48ba18872d11d434

Best Regards,
Michel.

mo5245, 2023-03-17:
Yes, builds now, also had to add .bat for the CMAKE_MAKE_PROGRAM in CMakeCache

`CMAKE_MAKE_PROGRAM:FILEPATH=C:/depot_tools/ninja.bat`

closed 2023-03-17 by mo5245