mpromonet/webrtc-streamer #568: Can you handle the problem of compiling under win10

JackyLucifer, 2023-03-11:
Under win10, I compile according to the steps of. github/workflow/cpp-windows.yml file

However, the current compilation fails. The prompt content is the same as that in the action of github. Please handle it

JackyLucifer, 2023-03-11:
[image]

mpromonet, 2023-03-11:
Hi @JackyLucifer 

I saw this, maintaining windows build is painful... I did not find a fix for now.
It seems a problem in shell interpretation, the single quote vs double quote ....
Don't hesitate to submit a pull request to fix it.

Best Regards,
Michel.

JackyLucifer, 2023-03-11:
Well, thank you very much.

mpromonet, 2023-03-11:
Hi @JackyLucifer 

Build on windows is fixed, docker on windows was built. There is till a problem in github action.

Best Regards,
Michel.

JackyLucifer, 2023-03-12:
[image]

JackyLucifer, 2023-03-13:
i see it is ok ，thank you very much.

closed 2023-03-13 by JackyLucifer