mpromonet/webrtc-streamer #571: 什么时候可以出一个支持H265的版本

liusuyi2021, 2023-04-01:
你好，我是中国的用户，我使用webrtc-streamer开发海康的相机进行视频播放，之前都是H264的编码，但是现在要改成H265,什么时候可以支持H265

mpromonet, 2023-04-01:
日本語が話せません

LK-5919, 2023-04-01:
Hello, I am a user in China. I use webrtc-streamer to develop Haikang's camera for video playback. Previously, it was coded in H264, but now it has to be changed to H265. When will this support H265??

mpromonet, 2023-04-01:
Hi,

To add support on rtsp backend may be achieve implementing H265 decoding. For a support on webrtc it need update on SDK that may be ongoing (see https://bugs.chromium.org/p/webrtc/issues/list?q=h265&can=2)

Best Regards,
Michel.

liusuyi2021, 2023-04-02:
sorry，My English level is limited。
我不太理解你说的是什么意思。我需要您能给我解释一下。

piaoyisong, 2023-04-27:
@liusuyi2021 请问播放 h265 问题解决了吗？

liusuyi2021, 2023-04-28:
没有，这个必须发布者解决

yangjieshao, 2023-06-20:
> sorry，My English level is limited。 我不太理解你说的是什么意思。我需要您能给我解释一下。

意思就是等谷歌支持 但是谷歌还在扯皮不愿意支持

mpromonet, 2023-07-08:
對不起，我的中文很有限

closed 2023-07-08 by mpromonet

yangjieshao, 2023-07-21:
Good news
好消息，快了快了
[image]

https://bugs.chromium.org/p/webrtc/issues/detail?id=13485#c28