mpromonet/webrtc-streamer #572: New Stream in config.json is not getting streamed

cloudties, 2023-04-01:
**Describe the bug**
Started the docker container by passing it via command line. In the config.json we added a new rtsp stream . In the webpage a new tab appear but the stream is not getting played 

**To Reproduce**
Steps to reproduce the behavior:
1. Add a new entry in config.json
2. Launched the docker in a Rasberry pi . May of the existing stream are getting stream properly . 
3. Connected a pi camera and even a USB camera to pi . Both tested separately and simultaneously 
4. On the web UI the new tab is appearing ,but not stream viewed
5. In order to check the camera funciton. we tried the libcamera module 
 libcamera-vid -t 0 --inline -o - | cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554/stream}' :demux=h264   --rtsp-host=192.168.29.63 
 The stream is getting generated and using VLC we are able to watch the rtsp stream from camera . But the same stream url added to the  config.json is not getting played 
6. All the devices are in the same network and accessible

**Expected behavior**
New working streams from camera should appear as tab and should show the live stream , like we have the sample 
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

mpromonet, 2023-04-02:
Hi @cloudties 

You may look to the log, it may helps you to understand the problem. Did you try to use network=host starting docker ?

Best Regards,
Michel.

cloudties, 2023-04-02:
> Hi @cloudties
> 
> You may look to the log, it may helps you to understand the problem. Did you try to use network=host starting docker ?
> 
> Best Regards, Michel.

Hi,

Wonderful ,that solved the issue . i get the stream on UI . 
How ever in the rtspstreamer i get a extra pattern in the url  "track1" . Is this automatically appended by the server ? 

handleCmd_SETUP:SETUP rtsp://192.168.29.63:8554/live/track1 RTSP/1.0
CSeq: 3
User-Agent: LIVE555 Streaming Media v2023.01.19
Transport: RTP/AVP/TCP;unicast;interleaved=0-1

Additionally , please add the network configuration in case of bridge or nat  for that docker . That helps 

Thanks Michel for the quick response .

Saju .

cloudties, 2023-04-06:
Closed after adding proper docker parameters

closed 2023-04-06 by cloudties