mpromonet/webrtc-streamer #580: Why can not using my server for public webrtc-streamer

Nguyen-Thanh-Tung, 2023-04-22:
Hi, I am facing an issue. I am using webrtc-streamer on my server (has turn server), it work with local network but it is not with other network. 
`
[
	{
		"candidate" : "candidate:1505818807 1 udp 2122260223 <myPrivateIp> 51100 typ host generation 0 ufrag cdDK network-id 1 network-cost 50",
		"sdpMLineIndex" : 0,
		"sdpMid" : "0"
	},
	{
		"candidate" : "candidate:2193873142 1 udp 1686052607 <myPublicIp> 51100 typ srflx raddr <myPrivateIp> rport 51100 generation 0 ufrag cdDK network-id 1 network-cost 50",
		"sdpMLineIndex" : 0,
		"sdpMid" : "0"
	}
]`
But when i deploy it to Google cloud or Amazon cloud, it work perfect. 
`[
	{
		"candidate" : "candidate:1005718130 1 udp 2122260223 <privateIpGCP> 60419 typ host generation 0 ufrag OuET network-id 1 network-cost 50",
		"sdpMLineIndex" : 0,
		"sdpMid" : "0"
	},
	{
		"candidate" : "candidate:2199625048 1 udp 1686052607 <publicIpGCP> 60419 typ srflx raddr <privateIpGCP> rport 60419 generation 0 ufrag OuET network-id 1 network-cost 50",
		"sdpMLineIndex" : 0,
		"sdpMid" : "0"
	},
	{
		"candidate" : "candidate:3458223196 1 udp 41885695 <privateIpGCP> 49580 typ relay raddr <publicIpGCP> rport 60419 generation 0 ufrag OuET network-id 1 network-cost 50",
		"sdpMLineIndex" : 0,
		"sdpMid" : "0"
	},
	{
		"candidate" : "candidate:3310935782 1 tcp 1518280447 <privateIpGCP> 50013 typ host tcptype passive generation 0 ufrag OuET network-id 1 network-cost 50",
		"sdpMLineIndex" : 0,
		"sdpMid" : "0"
	}
]`

I dont know the reason for this different. Please help me.

Nguyen-Thanh-Tung, 2023-04-22:
After some hours research about webrtc, to resolve my problem, i did some thing:
1. Open range port 49152-65535 with UDP for my server (min and max port can change if you want)
2. Run webrtc-streamer using param `-R 49152-65535`
3. Use trick with file webrtcstreamer.js:
Replace `
/*
* AJAX /getIceCandidate callback
*/
WebRtcStreamer.prototype.onReceiveCandidate = function(dataJson) {
	console.log("candidate: " + JSON.stringify(dataJson));
	if (dataJson) {
		for (var i=0; i<dataJson.length; i++) {
			var candidate = new RTCIceCandidate(dataJson[i]);
			
			console.log("Adding ICE candidate :" + JSON.stringify(candidate) );
			this.pc.addIceCandidate(candidate).then( () =>      { console.log ("addIceCandidate OK"); }
				, (error) => { console.log ("addIceCandidate error:" + JSON.stringify(error)); } );
		}
		this.pc.addIceCandidate();
	}
}
`
with
`
/*
   * AJAX /getIceCandidate callback
   */
  WebRtcStreamer.prototype.onReceiveCandidate = function (dataJson) {
    console.log("candidate: " + JSON.stringify(dataJson));
    if (dataJson) {
      for (let i = 0; i < dataJson.length; i++) {
        // trick to using public ip
        const tmp = {
          ...dataJson[i],
          candidate: dataJson[i].candidate.replace("<myPrivateIp>", "<myPublicIp>"),
        };
        const candidate = new RTCIceCandidate(tmp);

        console.log("Adding ICE candidate :" + JSON.stringify(candidate) );
        this.pc.addIceCandidate(candidate).then(
          () => {
            console.log ("addIceCandidate OK");
          },
          (error) => {
            console.log ("addIceCandidate error:" + JSON.stringify(error));
          }
        );
      }
      this.pc.addIceCandidate();
    }
  };

`

closed 2023-04-22 by Nguyen-Thanh-Tung

yeafel666, 2023-10-28:
Great, your answer is effective and has truly helped me!  thanks, bro .    god bless you !

yeafel666, 2023-10-28:
> After some hours research about webrtc, to resolve my problem, i did some thing:
> 
> 1. Open range port 49152-65535 with UDP for my server (min and max port can change if you want)
> 2. Run webrtc-streamer using param `-R 49152-65535`
> 3. Use trick with file webrtcstreamer.js:
>    Replace
>    `WebRtcStreamer.prototype.onReceiveCandidate = function(dataJson) { console.log("candidate: " + JSON.stringify(dataJson)); if (dataJson) { for (var i=0; i<dataJson.length; i++) { var candidate = new RTCIceCandidate(dataJson[i]); console.log("Adding ICE candidate :" + JSON.stringify(candidate) ); this.pc.addIceCandidate(candidate).then( () =>      { console.log ("addIceCandidate OK"); } , (error) => { console.log ("addIceCandidate error:" + JSON.stringify(error)); } ); } this.pc.addIceCandidate(); } }`
>    with
>    `WebRtcStreamer.prototype.onReceiveCandidate = function (dataJson) { console.log("candidate: " + JSON.stringify(dataJson)); if (dataJson) { for (let i = 0; i < dataJson.length; i++) { // trick to using public ip const tmp = { ...dataJson[i], candidate: dataJson[i].candidate.replace("<myPrivateIp>", "<myPublicIp>"), }; const candidate = new RTCIceCandidate(tmp); console.log("Adding ICE candidate :" + JSON.stringify(candidate) ); this.pc.addIceCandidate(candidate).then( () => { console.log ("addIceCandidate OK"); }, (error) => { console.log ("addIceCandidate error:" + JSON.stringify(error)); } ); } this.pc.addIceCandidate(); }   };`
> 
> Both using turn server and not using turn server, it work all

Great, your answer is effective and has truly helped me! thanks, bro . god bless you !

mpromonet, 2023-10-28:
Hi,

This is a strange approach to patch ice candidate. I guess ice candidate are not correctly discovered, or the turn server is not configured correctly, or the turn is not correctly used.

Best Regards,
Michel.