mpromonet/webrtc-streamer #584: cmake . failed. fatal error: 'string' file not found

iqbal-h, 2023-05-04:
**Describe the bug**
cmake . failed on fresh copy of webrtc-streamer and webrtc.
It use to build fine a while ago but now giving errors on std includes.

**To Reproduce**
The following steps are required to build the project, and will install the dependencies above:

Install the Chromium depot tools

pushd ..
git clone https://chromium.googlesource.com/chromium/tools/depot_tools.git
export PATH=$PATH:`realpath depot_tools`
popd
Download WebRTC

mkdir ../webrtc
pushd ../webrtc
fetch --no-history webrtc 
popd
Build WebRTC Streamer

cmake . && make

**Expected behavior**
Should build successfully

**Environment**
Compiler: g++ (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
System Architecture: x86_64
C++ Standard Libraries: 	libstdc++.so.6 (libc6,x86-64) => /lib/x86_64-linux-gnu/libstdc++.so.6
	libstdc++.so.6 (libc6) => /lib/i386-linux-gnu/libstdc++.so.6
	libstdc++.so.6 (libc6) => /lib32/libstdc++.so.6

**Additional context**
```
webrtc-streamer$ cmake .
-- The C compiler identification is GNU 11.3.0
-- The CXX compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMAKE_BUILD_TYPE = Release
WEBRTCROOT = /media/wisn/sec/webrtc-streamer/../webrtc
WEBRTCDESKTOPCAPTURE= ON
CMAKE_CXX_COMPILER_ID=GNU
-- Found Git: /usr/bin/git (found version "2.34.1") 
CMAKE_FIND_ROOT_PATH = 
ALSA_FOUND = FALSE
PulseAudio_FOUND = 0
GTK3_FOUND = 1
Done. Made 1295 targets from 257 files in 465ms
[1/3676] CXX obj/api/audio_options_api/audio_options.o
FAILED: obj/api/audio_options_api/audio_options.o 
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/api/audio_options_api/audio_options.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_OZONE=1 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_GNU_SOURCE -DCR_CLANG_REVISION=\"llvmorg-17-init-8029-g27f27d15-3\" -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_STRICT_FIELD_TRIALS=0 -DRTC_ENABLE_VP9 -DRTC_DAV1D_IN_INTERNAL_DECODER_FACTORY -DWEBRTC_HAVE_SCTP -DWEBRTC_USE_H264 -DWEBRTC_LIBRARY_IMPL -DWEBRTC_ENABLE_AVX2 -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -I../.. -Igen -I../../third_party/abseil-cpp -Wall -Wextra -Wimplicit-fallthrough -Wextra-semi -Wunreachable-code-aggressive -Wthread-safety -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-ignored-pragma-optimize -Wno-deprecated-builtins -Wno-bitfield-constant-conversion -Wno-deprecated-this-capture -Wshadow -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -ffp-contract=off -m64 -msse3 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -ffile-compilation-dir=. -no-canonical-prefixes -ftrivial-auto-var-init=pattern -O2 -fdata-sections -ffunction-sections -fno-unique-section-names -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wexit-time-destructors -Wglobal-constructors -Wno-shadow -Wctad-maybe-unsupported -Wc++11-narrowing -Wundef -Wunused-lambda-capture -std=c++20 -Wno-trigraphs -gsimple-template-names -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -c ../../api/audio_options.cc -o obj/api/audio_options_api/audio_options.o
In file included from ../../api/audio_options.cc:11:
../../api/audio_options.h:16:10: fatal error: 'string' file not found
#include <string>
         ^~~~~~~~
```

mpromonet, 2023-05-04:
Hi,

It seems stuff is missing, std::string is part of standard library... you may get information from https://webrtc.github.io/webrtc-org/native-code/development/prerequisite-sw/ or [Dockerfile](https://github.com/mpromonet/webrtc-streamer/blob/master/Dockerfile)

Best Regards,
Michel.

iqbal-h, 2023-05-08:
Thanks for sharing resources.
I tried installing pre-requisites using the shared links but it did not work.
Ultimately, a fresh OS worked.