mpromonet/webrtc-streamer #586: cmke error Compilation failed， What is the reason for this? Thank you

zhuya1996, 2023-05-08:
[image]
[image]
![QQ截图20230508213620](https://user-images.githubusercontent.com/25482513/236843349-92c721e9-23e7-42ad-871a-5a9ce1b0c28
[image]
2.png)

iqbal-h, 2023-05-08:
CMakeList.txt is inside webrtc-streamer directory.
`.../webrtc-streamer:~# cmake .`

zhuya1996, 2023-05-08:
> 

Put it in the depot_ Are tools in the same level directory？

zhuya1996, 2023-05-08:
> 目录应该是这样的：
> 
> ```
> .../home/depot_tools:~# 
> .../home/webrtc:~# 
> .../home/webrtc-streamer:~# 
> ```
> 
> 添加 depot_tools 的路径是必须的， `cmake . && make`只能在 webrtc-streamer 中运行

[image]
[image]

zhuya1996, 2023-05-08:
> > 目录应该是这样的：
> > ```
> > .../home/depot_tools:~# 
> > .../home/webrtc:~# 
> > .../home/webrtc-streamer:~# 
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 添加depot_tools的路径是必须的，`cmake . && make`只能在webrtc-streamer中运行
> 
> [image] [image]

There's still a problem

mpromonet, 2023-05-08:
Hi

Some dependencies are probably missing like glib. You should get needed environnent from dockerfile. 

Best Regards
Michel

zhuya1996, 2023-05-09:
o no，Not very good at using dockers

zhuya1996, 2023-05-09:
> 你好
> 
> 可能缺少一些依赖项，例如 glib。您应该从 dockerfile 获得所需的环境。
> 
> 最好的问候 米歇尔

[image]
Hello, have I compiled this successfully？？？

mpromonet, 2023-05-09:
Hi,

It looks like an executable, it is green, if you look to its permission, it should have +x permission.

Best Regards,
Michel.

zhuya1996, 2023-05-09:
> 你好，
> 
> 它看起来像一个可执行文件，它是绿色的，如果你查看它的权限，它应该有 +x 权限。
> 
> 最好的问候， 米歇尔。

[image]
After downloading, it is not an exe under Windows

mpromonet, 2023-05-09:
Hi,
Of course, you build a linux executable...
Best Regards
Michel

zhuya1996, 2023-05-10:
> 嗨， 当然，你构建了一个 linux 可执行文件...... 最好的问候 Michel

Hello, can you compile exe under Linux for use under Windows？

mpromonet, 2023-05-11:
Hi,

This need to cross compile for windows under linux, probably not an easy task to build webrtc lib. Why not build on windows for windows ?

Best Regards
Michel.

closed 2023-06-12 by mpromonet