mpromonet/webrtc-streamer #587: Why doesn't adding - o capture the desktop anymore? I want to capture the desktop and reduce CPU usage

zhuya1996, 2023-05-09:
Why doesn't adding - o capture the desktop anymore? I want to capture the desktop and reduce CPU usage

mpromonet, 2023-05-11:
Hi,

Using -o, only h264 is added to supported codec. Without it supports codec supported by webrtc, h264, vp8, ans others... You Can sée this in the /call answer.

Best Regards
Michel

zhuya1996, 2023-05-11:
> 你好，
> 
> 使用 -o，仅将 h264 添加到支持的编解码器。没有它支持 webrtc、h264、vp8 和其他支持的编解码器......你可以在 /call 答案中看到这个。
> 
> 最好的问候 米歇尔

I want to share video conference screens with you, but now the CPU takes up a large proportion. I want to use less than 3% of the CPU. Is it the only way to add gpu hard coding? - o supports rtsp, but does not support screen sharing？  thank you

zhuya1996, 2023-05-11:
> > 你好，
> > 使用-o，只将h264添加到支持的解码器。没有它支持webrtc、h264、vp8和其他支持的解码器......你可以在/call答案中看到这个。
> > 最好的问候米歇尔
> 
> 想和大家分享视频会议画面，但是现在CPU占用很大。我想使用少于 3% 的 CPU。它是添加gpu硬编码的唯一方法吗？- o 支持rtsp，但不支持屏幕共享？谢谢

Can desktop capture be forwarded directly using -0?